SICKAG/sick_scan_xd #389: How to change the device name by windows without ROS ?

sjeffreydong, 2024-09-19:
Dear 

How to change the device name by windows without ROS ? 
Please share the method.

rostest, 2024-09-19:
Thank you for your feedback. We recommend to use **SOPAS ET** or **SOPAS Air** to change the device name and store it in the EEPROM of the lidar. 

Alternatively you can use the SOPAS command `sWN LocationName` to set a device name:

[image]
See the telegram listing for details.

Without ROS, SOPAS commands can be sent using the `SickScanApiSendSOPAS` API function:

[image]
See the [API](https://github.com/SICKAG/sick_scan_xd/blob/develop/doc/sick_scan_api/sick_scan_api.md#diagnostic) description for details.

sjeffreydong, 2024-09-19:
I used SOPAS TO change the name but have the error : “status_code:2, status_message Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>, giving up after 1 unexpected answers.”
should i eidt the launch file?

closed 2024-11-06 by rostest