SICKAG/sick_scan_xd #390: Incorrect data is being published in the LaserScan data.

bandasaikrishna, 2024-09-19:
Hello,

I am using a PicoScan 150  lidar for a autonomous mobile navigation project in ROS2. 

ROS2 Version: Humble
OS: Ubuntu 22.04

I am using below command to launch the ros2 files related to PicoScan lidar.

`ros2 launch sick_scan_xd sick_picoscan.launch.py hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.2`

Connection is successful and I am able to see the point cloud data in rviz on "/cloud_all_fields_fullframe" topic with the fixed frame as "world" frame as shown in image below.

[image]

But when I try to visualize the laser scan data on "/scan_fullframe" or "/scan_segment" topics and by changing the fixed frame to "world_1"  , there are very few points visible in rviz as shown in below image.

[image]

Can some please help me to understand why I am facing problem with laser scan data while I am able to see correct data in point cloud ?

rostest, 2024-09-20:
Thank you for your feedback. The error is most likely caused by echo filtering. By default (i.e. parameter `"host_FREchoFilter"="2"` in the launchfile) only the last of 3 possible echos is expected from the lidar. If more echos are received, they will be filtered intensionally and only the last echo will be published in the laserscan topic with frame id "world_1". If there are scan points in other echos, they will not be published in the laserscan frame "world_1". Unlike laserscan messages, point cloud messages contain all echos. Therefore, rviz shows the complete scan in the point cloud topics, but not in the laserscan messages.

Please run the following steps:

* Enable all echoes by changing the parameter `host_FREchoFilter` to 1 (all echoes) in the launchfile: 
   ```
   <arg name="host_FREchoFilter" default="1" /> <!-- Optionally set FREchoFilter with 0 for FIRST_ECHO (default, EchoCount=1), 1 for ALL_ECHOS (EchoCount=3), or 2 for LAST_ECHO (EchoCount=1) -->
   ```
   With host_FREchoFilter=1, a laserscan message will be published for each echo with frame id "world_1_<echo_idx>". Set fixed frame in rviz to "world_1_0", "world_1_1" or "world_1_2" to visualize each echo:
   [image]
   [image]

* Alternatively, you can switch to the first echo by changing the `host_FREchoFilter' parameter in the launchfile to 0 (first echo): 
   ```
   <arg name="host_FREchoFilter" default="0" /> <!-- Optionally set FREchoFilter with 0 for FIRST_ECHO (default, EchoCount=1), 1 for ALL_ECHOS (EchoCount=3), or 2 for LAST_ECHO (EchoCount=1) -->
   ```
   Set the fixed frame in rviz to "world_1" to visualize the first echo:
   [image]

* Check for an updated firmware version with SOPAS ET or SOPAS Air and install the latest version.

bandasaikrishna, 2024-09-20:
Thank you for your quick response. Now  I am able to see correct LaserScan data on world_1 frame after setting host_FREchoFilter to 1.

closed 2024-09-20 by bandasaikrishna