SICKAG/sick_scan_xd #392: Support for MacOS

derivativerobmc, 2024-09-23:
Can the native non-ROS library be built and run on an ARM64 Mac? Are there any known issues with this?

We've already integrated the Windows version with our software, but would like to add support for our Mac version as well if possible.

rostest, 2024-09-24:
Thanks for your feedback. MacOS is not supported.

basicmonkey, 2024-09-24:
Is it that it's known not to work or that you won't support its application?

Since it's ethernet communication I can't see why it would be suitable for linux but not darwin. I could understand if there were physical device connections.

For transparency, I'm a new SICK customer and I have an entirely Mac computer inventory (other than servers).

Many thanks!

rostest, 2024-09-25:
Thanks for following up. sick_scan_xd has never been built or tested on MacOS. The sources are open source and you may give it a try. MacOS is not officially supported.

basicmonkey, 2024-09-25:
Many thanks! It's not my area of coding but might give it a go. Would be so valuable to my work!

closed 2024-10-17 by rostest