SICKAG/sick_scan_xd #394: Question: how to change fieldset

akashjinandra, 2024-10-08:
Hello, I am using ROS 1 Noetic, Ubuntu 20 and  SIck Tim 781, I am wondering how to change the fieldset. 
It looks like in RVIZ it defaults to 1 
[image]
However when i didn't see explicit instructions on what to send in order to change the fieldset. After digging around it seems like maybe with the ColaMsg service request it may be possible. But looking calling the service with this:
rosservice call /sick_tim_7xx/ColaMsg "{request: 'sRA LIDinputstate 2'}" 
However I get an error saying there is a mismatch:
[image]

rostest, 2024-10-08:
Thank you for your feedback. Use SOPAS ET to configure the fields and fieldset. We will take a closer look at the LIDinputstate telegrams.

akashjinandra, 2024-10-08:
when looking at the manual, it shows this: 
[image]
All the sRN commands seem to work, but none of the sWN commands work, we get an error:
[image]
And then our program hangs.

rostest, 2024-10-11:
Thanks for more information. The sick_scan_xd driver currently supports only the default fieldset selection mode, i.e. setting the active fieldset via digital inputs (FieldSetSelectionMethod=0). LIDinputstate telegrams are received by sick_scan_xd to monitor and update the active fieldset, but they are currently not published.  

The next release will improve the field evaluation for TiM7xx and TiM7xxS lidars:
* LIDinputstate telegrams will be published as ROS messages in topic /sick_tim_7xx/lidinputstate.
* FieldSetSelectionMethod and ActiveFieldSet options can be set in the launch file. By default the active field set can be set by digital inputs (FieldSetSelectionMethod=0). With the new option (FieldSetSelectionMethod=1) the active fieldset can be configured directly in the launchfile. Refer to the TiM 781 user manual and the telegram listing for full details of the fieldset options. Make sure that the latest firmware is installed on your lidar to use these options.

The next release is scheduled for publication within this month.

akashjinandra, 2024-10-11:
> Thanks for more information. The sick_scan_xd driver currently supports only the default fieldset selection mode, i.e. setting the active fieldset via digital inputs (FieldSetSelectionMethod=0). [...]

Hello, as you can see from my screenshot, the FieldSetSelection Method is already set to 1, I configured this through the SOPAS ET. 
[image]
I also already posted a screenshot of the TiM 781 User Manual in my posting, showing the commands, however in my screenshot, it did not work, when I literally copy and pasted the command. 
[image]

rostest, 2024-10-14:
Thanks for your reply. The upcoming release 3.6 will add two ros services to read and write the FieldSetSelectionMethod and ActiveFieldSet options. Please use the new services to read or change the fieldset configuration.

Example call for ROS1:
```
rosservice call /sick_tim_7xx/FieldSetRead "{}" # returns field_set_selection_method and active_field_set
rosservice call /sick_tim_7xx/FieldSetWrite "{field_set_selection_method_in: -1, active_field_set_in: 1}" # write active_field_set
```
Example call for ROS2:
```
ros2 service call /FieldSetRead sick_scan_xd/srv/FieldSetReadSrv "{}" # returns field_set_selection_method and active_field_set
ros2 service call /FieldSetWrite sick_scan_xd/srv/FieldSetWriteSrv "{field_set_selection_method_in: -1, active_field_set_in: 1}" # write active_field_set
```
Parameter summary:
* Parameter active_field_set < 0: do not overwrite, active_field_set > 0: index of active field otherwise (see operation manual for details about ActiveFieldSet telegram)
* Parameter field_set_selection_method < 0: do not overwrite, field_set_selection_method = 0: active field selection by digital inputs, field_set_selection_method = 1: active field selection by telegram

Initial values for field_set_selection_method and active_field_set can be configured in the launch file, e.g.:
```
<param name="active_field_set" type="int" value="-1"/> <!-- set ActiveFieldSet at startup: -1 = do not set (default), index of active field otherwise -->
<param name="field_set_selection_method" type="int" value="-1"/> <!-- set FieldSetSelectionMethod at startup: -1 = do not set (default), 0 = active field selection by digital inputs, 1 = active field selection by telegram -->
```
These options will be supported in the next sick_scan_xd release 3.6.

akashjinandra, 2024-10-14:
Hello and thank you for your reply. This should solve my problem, I would be willing to beta test the feature for you as well if you need any help.

akashjinandra, 2024-10-24:
> Thanks for your reply. The upcoming release 3.6 will add two ros services to read and write the FieldSetSelectionMethod and ActiveFieldSet options. Please use the new services to read or change the fieldset configuration.
> 
> Example call for ROS1:
> 
> ```
> rosservice call /sick_tim_7xx/FieldSetRead "{}" # returns field_set_selection_method and active_field_set
> rosservice call /sick_tim_7xx/FieldSetWrite "{field_set_selection_method_in: -1, active_field_set_in: 1}" # write active_field_set
> ```
> 
> Example call for ROS2:
> 
> ```
> ros2 service call /FieldSetRead sick_scan_xd/srv/FieldSetReadSrv "{}" # returns field_set_selection_method and active_field_set
> ros2 service call /FieldSetWrite sick_scan_xd/srv/FieldSetWriteSrv "{field_set_selection_method_in: -1, active_field_set_in: 1}" # write active_field_set
> ```
> 
> Parameter summary:
> 
> * Parameter active_field_set < 0: do not overwrite, active_field_set > 0: index of active field otherwise (see operation manual for details about ActiveFieldSet telegram)
> * Parameter field_set_selection_method < 0: do not overwrite, field_set_selection_method = 0: active field selection by digital inputs, field_set_selection_method = 1: active field selection by telegram
> 
> Initial values for field_set_selection_method and active_field_set can be configured in the launch file, e.g.:
> 
> ```
> <param name="active_field_set" type="int" value="-1"/> 
> <param name="field_set_selection_method" type="int" value="-1"/> 
> ```
> 
> These options will be supported in the next sick_scan_xd release 3.6.

In version 3.6.0, this seems to be working for me correctly. However it only works if I put the "-1" for the fieldset selection method. If I put "1" in the fieldset selection method it throws the same error as before "14 byte message ignored"

akashjinandra, 2024-11-06:
This feature is working for me on that branch you specified

anath93, 2024-12-06:
@rostest Thank you for this as I was looking for this feature too, works good. Is it possible to add a topic to see the Field state detection which is currently function of Digital output's ?

rostest, 2024-12-09:
@anath93 Thank you for your feedback. You can subscribe to the [LIDoutputstate](https://github.com/SICKAG/sick_scan_xd/blob/develop/msg/LIDoutputstateMsg.msg) messages to get the state of the digital outputs. LIDoutputstate messages are published in topic `/sick_tim_7xx/lidoutputstate`.

rostest, 2025-06-05:
Closed due to update to V. 3.7.0 (current develop)