SICKAG/sick_scan_xd #395: Not recieving lidinputstate topic

akashjinandra, 2024-10-08:
Hello, I am using ROS 1 Noetic, Ubuntu 20 and SIck Tim 781, I am wondering why I am not receiving the lidinputstate messages

when launching the node it looks successful:
[image]
However when I ping the node using rosnode info, it doesn't show up, only the outputstates do: 
[image]

rostest, 2024-10-08:
Thank you for your feedback. We will take a closer look at the LIDinputstate telegrams.

rostest, 2024-10-11:
LIDinputstate telegrams are received by sick_scan_xd to monitor and update the active fieldset. They are currently not published. 

The next release will improve the field evaluation for TiM7xx and TiM7xxS lidars:
* LIDinputstate telegrams will be published as ROS messages in topic /sick_tim_7xx/lidinputstate.
* FieldSetSelectionMethod and ActiveFieldSet options can be set in the launch file. By default the active field set can be set by digital inputs (FieldSetSelectionMethod=0). With the new option (FieldSetSelectionMethod=1) the active fieldset can be configured directly in the launchfile. Refer to the TiM 781 user manual and the telegram listing for full details of the fieldset options. Make sure that the latest firmware is installed on your lidar to use these options.

The next release is scheduled for publication within this month.