SICKAG/sick_scan_xd #396: Issue when compiling for ARM64

pb-64, 2024-10-08:
I'm trying to compile the library libsick_scan_xd_shared_lib.so for ARM64 with this comands:

mkdir -p ./build
pushd ./build
rm -rf ./*
export ROS_VERSION=0
cmake -DROS_VERSION=0 -DCMAKE_CXX_COMPILER=aarch64-linux-gnu-g++ -DLDMRS=0 -G "Unix Makefiles" ../sick_scan_xd

but I get these errors:

_italcadmin@LNX-Automation:~/sick_scan_ws/sick_scan_xd$ make -j4
Consolidate compiler generated dependencies of target sick_scan_xd_api_test
Consolidate compiler generated dependencies of target sick_scan_xd_api_dockertest
[  1%] Linking CXX executable sick_scan_xd_api_test
[  2%] Linking CXX executable sick_scan_xd_api_dockertest
Consolidate compiler generated dependencies of target sick_scan_xd_lib
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_test.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_test.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_test.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_test.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_test.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_test.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_test.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
Consolidate compiler generated dependencies of target sick_scan_xd_shared_lib
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_dockertest.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_dockertest.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_dockertest.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_dockertest.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_dockertest.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_dockertest.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_api_dockertest.dir/test/src/sick_scan_xd_api/sick_scan_xd_api_wrapper.c.o: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sick_scan_xd_api_test.dir/build.make:145: sick_scan_xd_api_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:169: CMakeFiles/sick_scan_xd_api_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/sick_scan_xd_api_dockertest.dir/build.make:129: sick_scan_xd_api_dockertest] Error 1
make[1]: *** [CMakeFiles/Makefile2:195: CMakeFiles/sick_scan_xd_api_dockertest.dir/all] Error 2
[ 48%] Built target sick_scan_xd_lib
[ 49%] Linking CXX shared library libsick_scan_xd_shared_lib.so
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_shared_lib.dir/roswrap/src/getopt/getopt.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_shared_lib.dir/roswrap/src/getopt/getopt.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_shared_lib.dir/roswrap/src/getopt/getopt.c.o: Relocations in generic ELF (EM: 62)
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: CMakeFiles/sick_scan_xd_shared_lib.dir/roswrap/src/getopt/getopt.c.o: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sick_scan_xd_shared_lib.dir/build.make:1025: libsick_scan_xd_shared_lib.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:117: CMakeFiles/sick_scan_xd_shared_lib.dir/all] Error 2
make: *** [Makefile:136: all] Error 2_

rostest, 2024-10-09:
Thank you for your feedback. Cross compilation is very system dependent, we generally recommend native build and compilation on the target whenever possible.

sick_scan_xd provides a C interface, so some sources like sick_scan_xd_api_wrapper.c are pure C. This may cause build errors if compiled with g++ instead of gcc. If cross compiling is required, try to specify both gcc and g++, e.g. `cmake -DROS_VERSION=0 -DCMAKE_C_COMPILER=aarch64-linux-gnu-gcc -DCMAKE_CXX_COMPILER=aarch64-linux-gnu-g++ -DLDMRS=0 -G "Unix Makefiles" ../sick_scan_xd`.

pb-64, 2024-10-09:
Thanks for the reply. 
I was able to compile both the library and the application for ARM64 but once transmit to the device if I launch the application I get this error:
./minimum_sick_scan_api_client: /lib/libc.so.6: version `GLIBC_2.34' not found (required by ./minimum_sick_scan_api_client).
I don't have the possibility to update the library on the device.
Is there any possibility to make what I compiled work on the target system too?

rostest, 2024-10-10:
Thanks for the information. My guess is that the cross compiler does not exactly match the system libraries or their versions installed on your target. If you need to cross compile, make sure your cross build matches the libraries on your target exactly. If you have a chance to install gcc and cmake on your target, please do so. You can then build the sick_scan_xd library and application directly on the target.