SICKAG/sick_scan_xd #399: Sick PicoScan150 "add_transform_xyz_rpy" not working

ravikiransama, 2024-10-15:
Hi,
I have an issue with sick pico scan 150. The "add_transform_xyz_rpy" is not working. The moment i enter non zero values, no point clouds are published on that topic in rviz. I use ros2 humble and sick_scan_xd humble version. I had built the sick scan from source.

Here are my launch and config files.
`<?xml version="1.0"?>
<launch>

    
    
    <arg name="hostname" default="192.168.0.101"/>  
    <arg name="udp_receiver_ip" default="192.168.0.100"/>        
    <arg name="nodename" default="sick_picoscan1"/>
    <arg name="publish_frame_id" default="world" />                                 
    <arg name="publish_laserscan_segment_topic" default="scan1_segment" />           
    <arg name="publish_laserscan_fullframe_topic" default="scan1_fullframe" />       
    <arg name="imu_topic" default="imu" />                                          
    <arg name="add_transform_xyz_rpy" default="0,0,0,0,0,0"/>
    <arg name="scandataformat" default="2"/>                  
    <arg name="performanceprofilenumber" default="-1"/>       
    <arg name="all_segments_min_deg" default="-138.0" />      
    <arg name="all_segments_max_deg" default="+138.0" />      

    <arg name="host_FREchoFilter" default="2" />              

    <arg name="custom_pointclouds" default="cloud_unstructured_segments cloud_unstructured_fullframe cloud_polar_unstructured_segments cloud_polar_unstructured_fullframe cloud_all_fields_fullframe"/> 
    <node name="$(arg nodename)" pkg="sick_scan_xd" type="sick_generic_caller" respawn="false" output="screen" required="true">    
        <param name="scanner_type" type="string" value="sick_picoscan"/>

        
        <param name="hostname" type="string" value="$(arg hostname)" />                     
        <param name="port" type="int" value="2115" />                                       
        <param name="udp_receiver_ip" type="string" value="$(arg udp_receiver_ip)" />       

        
        <param name="udp_sender" type="string" value="" />                                  
        <param name="udp_port" type="int" value="2115" />                                   
        <param name="all_segments_min_deg" type="double" value="$(arg all_segments_min_deg)" />              
        <param name="all_segments_max_deg" type="double" value="$(arg all_segments_max_deg)" />              
        <param name="publish_frame_id" type="string" value="$(arg publish_frame_id)" />     
        <param name="publish_laserscan_segment_topic" type="string" value="$(arg publish_laserscan_segment_topic)" />           
        <param name="publish_laserscan_fullframe_topic" type="string" value="$(arg publish_laserscan_fullframe_topic)" />       
        <param name="udp_input_fifolength" type="int" value="20" />                         
        <param name="msgpack_output_fifolength" type="int" value="20" />                    
        <param name="verbose_level" type="int" value="1" />                                 
        <param name="measure_timing" type="bool" value="True" />                            
        <param name="export_csv" type="bool" value="False" />                               
        <param name="export_udp_msg" type="bool" value="False" />                           
        <param name="logfolder" type="string" value="" />                                   
        <param name="send_udp_start" type="bool" value="False" />                           
        <param name="send_udp_start_string" type="string" value="magicalActivate" />        
        <param name="udp_timeout_ms" type="int" value="60000" />                            
        <param name="scandataformat" type="int" value="$(arg scandataformat)" />            
        <param name="performanceprofilenumber" type="int" value="$(arg performanceprofilenumber)"/> 
        
        <param name="imu_enable" type="bool" value="True"/>                                 
        <param name="imu_udp_port" type="int" value="7503"/>                                
        <param name="imu_latency_microsec" type="int" value="0"/>                           
        <param name="imu_topic" type="string" value="$(arg imu_topic)"/>                    
        
        
        
         
        
        
        
        <param name="add_transform_xyz_rpy" type="string" value="0.398 -0.213 0.12,0,0,-0.785398" /> 

        
        <param name="sopas_tcp_port" type="string" value="2111" />                          
        <param name="start_sopas_service" type="bool" value="True" />                       
[... 289 lines cut ...]

        
        <param name="cloud_structured_hires1_segments" type="string" value="coordinateNotation=3 updateMethod=1 fields=x,y,z,i,range,azimuth,elevation echos=0,1,2 layers=14 reflectors=0,1 infringed=0,1 rangeFilter=0,999,0 topic=/cloud_structured_hires1_segments frameid=world publish=1"/>

        
        <param name="cloud_structured" type="string" value="coordinateNotation=3 updateMethod=0 fields=x,y,z,i,range,azimuth,elevation echos=0,1,2 layers=1,2,3,4,5,7,8,9,10,11,12,13,15,16 reflectors=0,1 infringed=0,1 rangeFilter=0,999,0 topic=/cloud_structured frameid=world publish=1"/>

        
        <param name="cloud_structured_segments" type="string" value="coordinateNotation=3 updateMethod=1 fields=x,y,z,i,range,azimuth,elevation echos=0,1,2 layers=1,2,3,4,5,7,8,9,10,11,12,13,15,16 reflectors=0,1 infringed=0,1 rangeFilter=0,999,0 topic=/cloud_structured_segments frameid=world publish=1"/>

        
        <param name="cloud_all_fields_segments" type="string" value="coordinateNotation=3 updateMethod=1 fields=x,y,z,i,range,azimuth,elevation,layer,echo,reflector echos=0,1,2 layers=1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16 reflectors=0,1 infringed=0,1 rangeFilter=0,999,0 topic=/cloud_all_fields_segments frameid=world publish=1"/>

        
        <param name="cloud_all_fields_fullframe" type="string" value="coordinateNotation=3 updateMethod=0 fields=x,y,z,i,range,azimuth,elevation,layer,echo,reflector echos=0,1,2 layers=1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16 reflectors=0,1 infringed=0,1 rangeFilter=0,999,0 topic=/cloud_all_fields_fullframe frameid=world publish=1"/>

        
        <param name="tf_base_frame_id" type="string" value="map" />              
        <param name="tf_base_lidar_xyz_rpy" type="string" value="0,0,0,0,0,0" /> 
        <param name="tf_publish_rate" type="double" value="10" />                

    </node>

</launch>
`

rostest, 2024-10-16:
Thank you for your feedback. Due to a configuration issue, the full frame point clouds are not published if the configured rotation is only about the z axis (i.e. roll=0, pitch=0, yaw!=0). Please use the update provided in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2407xx (driver/src/sick_scansegment_xd/config.cpp modified), rebuild and run again.

Note that the parameter "add_transform_xyz_rpy" expects 6 comma separated values. Please replace `<param name="add_transform_xyz_rpy" type="string" value="0.398 -0.213 0.12,0,0,-0.785398" />` (missing commas) with `<param name="add_transform_xyz_rpy" type="string" value="0.398,-0.213,0.12,0,0,-0.785398" />` (or any other transform you want to apply) in your launch file.

ravikiransama, 2024-10-16:
Thank you for the suggestion. It works now with the update in the branch. Regarding the "add_transform_xyz_rpy" , it was typo mistake. The api now publishes the correctly. Another question, do you have any suggestion for merging 2 laser scanners? For some reason ira laser tools is not working with the sick scanners - pico100.

closed 2024-10-16 by ravikiransama