SICKAG/sick_scan_xd #400: Driver crashes reporting an error code -11

Navsea, 2024-10-19:
Sporadically it happens that when operating, so we're receiving point clouds, that we suddenly don't receive any point clouds anymore.
The linux kernel reports an error code of -11.

I would copy the printout here, but for some reason I cannot copy paste it here, nor upload an image of it...

This would indicate a segmentation fault, memory access violation.

Is this a known issue?

rostest, 2024-10-21:
Thank you for your feedback. This is not a known issue. What lidar are you using? Are you using ROS-1, ROS-2 or the sick_scan_xd API?

The error may be related to issue #385 (debug assertion). Please update with branch [feature/issue_update_2407xx](https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2407xx) and try again. 

If the problem persists, please build with debug information (cmake flag `-DO=0`), run sick_scan_xd in gdb (using `launch-prefix="gdb -ex run --args"` in the ROS launch file or `gdb --args ./build/sick_scan_xd_api_test <launchfile>` for the API) and send a backtrace (command `bt` in gdb) for further analysis.

Navsea, 2024-10-22:
Hi,

We're using the Multiscan 136, ROS2 humble and the corresponding ROS driver.
We'll try updating the driver to see if that fixes the issue, but we're currently in the field and returning back tomorrow, so we might nog report back immediately...

rostest, 2024-11-14:
@Navsea Thanks for following up. I am closing this thread for now, but please feel free to reopen it if you have any new information or further issues.

closed 2024-11-14 by rostest