SICKAG/sick_scan_xd #401: Problems compiling ROS1 Version for Kinetic [Ubuntu 16.04]

CTech-dev0, 2024-10-19:
Hi:

I'm trying to compile the library in our Ubuntu 16.04 machine (Kernel version 4.15.0-112-generic). When invoking catkin_isolated (as show in the ROS1 tutorial), the compilation process goes well upto 60% aproximately and then it fails to compile , showing the following errrors:

`In file included from /home/ct-dev/sick_scan_ws/src/sick_scan_xd/driver/src/sick_cloud_transform.cpp:58:0:`

`In file included from /home/ct-dev/sick_scan_ws/src/sick_scan_xd/driver/src/sick_cloud_transform.cpp:58:0:
/home/ct-dev/sick_scan_ws/src/sick_scan_xd/include/sick_scan/sick_cloud_transform.h:157:51: error: array must be initialized with a brace-enclosed initializer
         Vector3D m_translation_vector = { 0, 0, 0 };                  // translational part x,y,z of the 6D pose, default: 0`

`/home/ct-dev/sick_scan_ws/src/sick_scan_xd/include/sick_scan/sick_cloud_transform.h:157:51: error: too many initializers for ‘sick_scan_xd::SickCloudTransform::Vector3D {aka std::array<float, 3ul>}’`

`/home/ct-dev/sick_scan_ws/src/sick_scan_xd/include/sick_scan/sick_cloud_transform.h:158:67: error: array must be initialized with a brace-enclosed initializer
         Matrix3x3 m_rotation_matrix = { 1, 0, 0, 0, 1, 0, 0, 0, 1 };  // rotational part roll,pitch,yaw by 3x3 rotation matrix, default: 3x3 identity`

`/home/ct-dev/sick_scan_ws/src/sick_scan_xd/include/sick_scan/sick_cloud_transform.h:158:67: error: too many initializers for ‘sick_scan_xd::SickCloudTransform::Matrix3x3 {aka std::array<std::array<float, 3ul>, 3ul>}’`

`In file included from /home/ct-dev/sick_scan_ws/src/sick_scan_xd/include/sick_scan/sick_scan_common.h:79:0,
                 from /home/ct-dev/sick_scan_ws/src/sick_scan_xd/driver/src/sick_generic_field_mon.cpp:60:`

`/home/ct-dev/sick_scan_ws/src/sick_scan_xd/include/sick_scan/sick_cloud_transform.h:157:51: error: array must be initialized with a brace-enclosed initializer
         Vector3D m_translation_vector = { 0, 0, 0 };                  // translational part x,y,z of the 6D pose, default: 0`

`/home/ct-dev/sick_scan_ws/src/sick_scan_xd/include/sick_scan/sick_cloud_transform.h:157:51: error: too many initializers for ‘sick_scan_xd::SickCloudTransform::Vector3D {aka std::array<float, 3ul>}’`

`/home/ct-dev/sick_scan_ws/src/sick_scan_xd/include/sick_scan/sick_cloud_transform.h:158:67: error: array must be initialized with a brace-enclosed initializer
         Matrix3x3 m_rotation_matrix = { 1, 0, 0, 0, 1, 0, 0, 0, 1 };  // rotational part roll,pitch,yaw by 3x3 rotation matrix, default: 3x3 identity`

....

Here is a picture attached to show the actual shell outputs.

[image]

At the end of the error message it shows the following:

`<== Failed to process package 'sick_scan_xd': 
  Command '['make', '-j4', '-l4']' returned non-zero exit status 2`

Has anyone encountered the same problem before ?.

rostest, 2024-10-21:
Thank you for your feedback. Ubuntu 16 is no longer supported. The sick_scan_xd driver requires C++ 14. Please upgrade to at least Ubuntu 20 or newer with gcc version >= 5.

CTech-dev0, 2024-10-21:
Thanks for the quick response.
We will adapt to the current requirements.

Regards.

closed 2024-10-21 by CTech-dev0