SICKAG/sick_scan_xd #403: Encoder Data Not Published for LMS4000 in ROS2

asokolowski-it, 2024-10-20:
I am using an LMS4000 lidar sensor with the sick_scan_xd driver in ROS2, and while the point cloud and other data are being successfully published, I am unable to receive any encoder data on the /sick_lms_4xxx/encoder topic.

Steps Taken:
LMS4000 Firmware & Software Updates:

- I have performed both firmware and software updates on the LMS4000 to the latest available versions.

ROS2 Environment Setup:

- I am running ROS2 Humble.

- I used the following launch command: ros2 launch sick_scan_xd sick_lms_4xxx.launch.py hostname:=192.168.2.100 encoder_mode:=2 (also 1, 3, 4).

Point Cloud Data:

- The /cloud topic publishes data at around 600 Hz as expected, confirming the lidar is functional in other respects.

Encoder Topic:

- When attempting to listen to the /sick_lms_4xxx/encoder topic using ros2 topic echo /sick_lms_4xxx/encoder, I receive no data.

- Checking the publication rate using ros2 topic hz /sick_lms_4xxx/encoder also shows no messages being published.

Branch and Build:

- 3.5.0

Additional Information:

- The encoder works and shows values in SOPAS ET, indicating the hardware and wiring are correct, but the data is not available in ROS2.

rostest, 2024-10-21:
Thank you for your feedback. Please send a pcapng file with scan and encoder data from your lidar for further analysis:
* Start wireshark and select the interface used for TCP communication with the lidar.
* Start sick_scan_xd with encoder_mode:=2
* Wait a few seconds for rviz to display the point cloud
* Save the network traffic to a pcapng file (Wireshark menu -> File -> Save -> pcapng file)

Please provide this pcapng file for further analysis.

asokolowski-it, 2024-10-21:
Thank you for your feedback. As requested, I’ve captured the network traffic between my ROS 2 system and the LMS4000 lidar while running the sick_scan_xd driver with encoder_mode set to 2.

Steps I followed:

- I started Wireshark and selected the network interface used for TCP communication with the lidar (IP: 192.168.2.100).

- I let the driver run for several seconds, ensuring that the point cloud data was visible in RViz.

- I captured the network traffic and saved it to a pcapng file.

- I’ve attached the pcapng file below for further analysis: 
[lms4000.zip](https://github.com/user-attachments/files/17463419/lms4000.zip)

[image]

Please let me know if you need any additional information or further adjustments. Thank you for your assistance!

asokolowski-it, 2024-10-22:
Thank you very much for the update and the guidance! I appreciate your help in addressing this issue.

feature/issue_update_2407xx working as expected. I checked ros2 topic echo /sick_lms_4xxx/encoder, and I can now see the expected encoder messages being published.
[image]

As requested, I’ve attached the new pcapng if you still need it. You can download it using the following link: 
[wireshark.zip](https://github.com/user-attachments/files/17478700/wireshark.zip)

closed 2024-10-22 by asokolowski-it

rostest, 2024-10-22:
@asokolowski-it Thanks for the good teamwork!