SICKAG/sick_scan_xd #407: VS2022 - c++ minimum_sick_scan_api_client  example issue

Nigel-BMV, 2024-10-24:
The Project has been build for the new 3.6.0. release according to the instructions.

The generic_caller and test executable correctly connect with the Picoscan device (w. IP 10.10.10.100), which returns data in the CMD.

I am now attempting to get the minimum  example project functional; but am running into an issue when I attempt to run it in debug mode.
Namely the line : 
**_SickScanApiLoadLibrary(sick_scan_api_lib);_**

After which the project throws an error relating to the **_SickScanApiCreate_** call.
[image]
This seems to be a nullpointer exception?

I have attempted to place the compiled sick_scan_xd_shared_lib.dll in the executable folder, but the error remains the same with or without it present.

rostest, 2024-10-25:
Thanks for your feedback. I cannot reproduce the error, the minimum_sick_scan_api_client example runs in debug mode with both VS2019 and VS2022.

The reason for the exception is currently unclear. Did you use the same compiler settings for both the sick_scan_xd library and the minimum_sick_scan_api_client example with cmake and the provided CMakeLists.txt files? Can you see the stack trace by clicking on "Show Call Stack" in the debugger? If so, please post a screenshot. 

According to the screenshot, msvcp140.dll is being loaded when the exception occurs. In debug mode I would expect the debug version of this dll, i.e. msvcp140d.dll instead of msvcp140.dll. It is recommended to use identical compiler settings for the exe and dll files. For example, you can simply copy all files from sick_scan_xd\build\Debug to sick_scan_xd\examples\cpp\Debug. After that the debug version of all libraries including msvcp140d.dll should be loaded at runtime.

Nigel-BMV, 2024-10-28:
> Thanks for your feedback. I cannot reproduce the error, the minimum_sick_scan_api_client example runs in debug mode with both VS2019 and VS2022.
> 
> The reason for the exception is currently unclear. Did you use the same compiler settings for both the sick_scan_xd library and the minimum_sick_scan_api_client example with cmake and the provided CMakeLists.txt files? Can you see the stack trace and function parameters by clicking on "Show Call Stack" in the debugger? If so, please post a screenshot.
> 
> According to the screenshot, msvcp140.dll is being loaded when the exception occurs. In debug mode I would expect the debug version of this dll, i.e. msvcp140d.dll instead of msvcp140.dll. It is recommended to use identical compiler settings for the exe and dll files. For example, you can simply copy all files from sick_scan_xd\build\Debug to sick_scan_xd\examples\cpp\Debug. After that the debug version of all libraries including msvcp140d.dll should be loaded at runtime.

[image]

However, copying the files from 'sick_scan_xd\build\Debug' was successful.

I am now, however, running into a different issue.
[image]

I have tried setting the command line arguments similarly to the picoscan start parameters with
'sick_picoscan.launch hostname:=<10.10.10.100> udp_receiver_ip:=<10.10.10.13>'

However, this also doesn't work.

I have no idea what to actually pass as argc and argv to 'SickScanApiInitByCli' at this point.

This will be a necessary input as we are intending to have multiple Picoscans in the same area and knowing which one is opened where, is a crucial step for that.

rostest, 2024-10-28:
Thanks for your reply. Function SickScanApiInitByCli expects (argc, argv) arguments as provided from the command line:
```
sick_scan_xd_api_test <launchfile> hostname:=<lidar-ip-address> udp_receiver_ip:=<pc-ip-address>
```
Example:
```
minimum_sick_scan_api_client launch/sick_picoscan.launch hostname:=10.10.10.100 udp_receiver_ip:=10.10.10.13
```

To call SickScanApiInitByCli directly with these arguments, you can provide (argc, argv) like this
```
char* argv[] = { "minimum_sick_scan_api_client", "launch/sick_picoscan.launch", "hostname:=10.10.10.100", "udp_receiver_ip:=10.10.10.13" }; 
int argc = 4;
SickScanApiHandle apiHandle = SickScanApiCreate(argc, &argv[0]);
SickScanApiInitByCli(apiHandle, argc, &argv[0]);
```

Adapt the path to the launch file if required. See https://github.com/SICKAG/sick_scan_xd?tab=readme-ov-file#generic-driver-api for details and examples.