SICKAG/sick_scan_xd #409: examples minimum_sick_scan_api_client library not found

benedettisamuele, 2024-10-25:
Hello, I compiled the library libsick_scan_xd_shared_lib.so for ARM64 with the following commands:
- mkdir -p ./build
- pushd ./build
- rm -rf ./*
- export ROS_VERSION=0
- cmake -DROS_VERSION=0 -DCMAKE_C_COMPILER=aarch64-linux-gnu-gcc -DCMAKE_CXX_COMPILER=aarch64-linux-gnu-g++ -DLDMRS=0 -G "Unix Makefiles" ../sick_scan_xd
- make -j4
- sudo make -j4 install 

Then, I compiled the examples/cpp minimum_sick_scan_api_client with:
- cmake -DCMAKE_C_COMPILER=aarch64-linux-gnu-gcc -DCMAKE_CXX_COMPILER=aarch64-linux-gnu-g++ -G "Unix Makefiles" ..
- make -j4

I transferred both files on my ARM64 device, put both files in the same folder but I get:
## WARNING SickScanApiLoadLibrary: LoadLibrary("libsick_scan_xd_shared_lib.so") failed

With sick_scan_xd_api_test instead of the minimum_sick_scan_api_client it works.

What is it wrong with my procedure?

Thanks

rostest, 2024-10-25:
Thanks for your feedback. There is nothing wrong with your procedure. Just make sure that libsick_scan_xd_shared_lib.so can be found in a folder included in your LD_LIBRARY_PATH.

The example minimum_sick_scan_api_client.cpp simply loads "libsick_scan_xd_shared_lib.so", while sick_scan_xd_api_test.cpp in addition looks for "libsick_scan_xd_shared_lib.so" in the folders "", "build/", "build_linux/", "src/build/", "src/build_linux/", "src/sick_scan_xd/build/", "src/sick_scan_xd/build_linux/", "./", "../". That's why LoadLibrary succeeds in sick_scan_xd_api_test.cpp, but not in minimum_sick_scan_api_client.cpp.

Copy libsick_scan_xd_shared_lib.so to a folder included in LD_LIBRARY_PATH on your target. Alternatively copy libsick_scan_xd_shared_lib.so, minimum_sick_scan_api_client and sick_scan_xd_api_test into the same directory and include this folder in LD_LIBRARY_PATH by ``export LD_LIBRARY_PATH=.:$LD_LIBRARY_PATH`` or  ``export LD_LIBRARY_PATH=`(pwd)`:$LD_LIBRARY_PATH``.

benedettisamuele, 2024-10-25:
Thanks for the quick reply.
It still doesn't work; on my ARM device I have the following folder tree, with library, launch and minimum application in the same folder:

[image]

Adding the LD_LIBRARY_PATH I have the following but still no success in loading the library:

[image]

I tried also to copy the for cycle of the loading library part from sick_scan_xd_api_test.cpp to the minimum_sick_scan_api_client.cpp: the load of the library fails again for all the folders taken into account:

[image]

Is something still missing?

Thanks a lot

rostest, 2024-10-28:
Thanks for your reply. This looks like a cross compilation issue to me. Loading the sick_scan_xd library is the same in minimum_sick_scan_api_client and sick_scan_xd_api_test (except for the paths). sick_scan_xd_api_test also links against pthread; you can try linking minimum_sick_scan_api_client with pthread by adding the following in examples/cpp/CMakeLists.txt
```
target_link_libraries(${PROJECT_NAME} "pthread")
target_link_options(${PROJECT_NAME} PUBLIC "LINKER:--no-as-needed")
```

Please first build and run minimum_sick_scan_api_client and sick_scan_xd_api_test without cross compiling on your Linux PC or target. If this works as expected, it is most likely a cross compilation problem. Note that cross compiling is very system dependent and is not officially supported.

closed 2024-11-14 by rostest