SICKAG/sick_scan_xd #411: Use ROS Time instead of sensor time

reAgriRobot, 2024-10-28:
Hi, 

I want to use the ROS2 time in a system without PTP/SNTP support. Is there a configuration parameter to support this? 

Thanks

reAgriRobot, 2024-11-14:
I have a MultiScan 136 connected. Without PTP, the header.stamp.sec value was 0, so I enabled PTP to obtain the correct timestamp. While this generally works, sometimes the value unexpectedly jumps to -1, causing system failures. This issue can occur more than three times in a day.

rostest, 2024-11-14:
Thanks for your reply. I cannot reproduce an invalid timestamp header.stamp.sec = 0 without PTP. Which topic or topics are you using? Are you using the latest sick_scan_xd release?
As the time stamp jumps irregularly from time to time when using PTP, this may be caused by network faults. Please check for missing PTP telegrams, network overload or lost packets when this error occurs.

closed 2024-12-02 by rostest

reinbja, 2025-02-25:
@reAgriRobot 
Was the issue fixed after an update to the latest release version of the driver? Maybe this issue #441 correlates to your behavior.