SICKAG/sick_scan_xd #413: Can't compile with VS2022

Janisthier, 2024-11-03:
i have the same issue as in the Thread 179. In VS2022 i get the error 1017. I've tried VS2019 too, but it isnt work ether.

Ive tried the description from the thread 179 but the CL put out an another error:  

[image]

rostest, 2024-11-04:
Thanks for your feedback. #179 refers to an older version. Please update sick_scan_xd to the latest release 3.6.0 with integrated msgpack11 and follow the [build instructions](https://github.com/SICKAG/sick_scan_xd?tab=readme-ov-file#build-on-windows-generic-without-ros):

[image]

Janisthier, 2024-11-09:
Thanks! now its working.

closed 2024-11-11 by rostest