SICKAG/sick_scan_xd #414: hello! build Error!!

wanilly, 2024-11-05:
[image]

You see the picture, the terminal print error message( error: array must be initialized with a brace-enclosed initializer )

I want to solve this problem... plz...

rostest, 2024-11-05:
Thanks for your feedback. The sick_scan_xd driver requires C++14. C++11 introduced the initialisation of a [std::array](https://en.cppreference.com/w/cpp/container/array) with braces like `std::array<int, 3> a = {1, 2, 3}`.

What gcc version and OS are you using? It looks like gcc does not yet support C++14, similar to #401 or #251. We recommend upgrading to Ubuntu >= 20.04 with gcc >= 5.

wanilly, 2024-11-06:
[image]
Unknown CMake command "target_link_options". 

gcc version:  5.4.0
ubuntu version: 16.04 LTS

wanilly, 2024-11-06:
Hello! I have a question!!... 

I use TIM7xxS object. I'm trying to port separately without using api ( without ROS and Ubuntu ). How can I access it? We need to use 18.04 version.

rostest, 2024-11-06:
Thank you for more information. Ubuntu 16.04 is no longer supported. If you are using the Linux subsystem on Windows, we recommend to upgrade to Ubuntu 20 or newer. The cmake command target_link_options requires cmake 3.13 or later; we will update the cmake requirements.

To use a TiM781S without the sick_scan_xd API or ROS, you need to establish a TCP connection and communicate with the lidar using the appropriate SOPAS commands. See the [TiM781S user manual](https://cdn.sick.com/media/docs/4/54/254/operating_instructions_tim7xxs_en_im0083254.pdf) and the [TiM781S telegram listing](https://cdn.sick.com/media/docs/7/27/927/telegram_listing_telegram_listing_ranging_sensors_lms1xx_lms5xx_tim2xx_tim5xx_tim7xx_lms1000_mrs1000_mrs6000_nav310_ld_oem15xx_ld_lrs36xx_lms4000_lrs4000_multiscan100_picoscan100_en_im0045927.pdf) for details. We recommend using the sick_scan_xd driver on Windows or Linux with Ubuntu 20 or newer.

wanilly, 2024-11-09:
Oh, I'm seokwan! Thank you for your response.

I have a one more question. If i use api, where i find Lidar
measurement data?? I need to your guide. I read document. But, I don't find
measurement data...

Thank you...

-- 
*Kim Seok Wan*
Handong Global University
Heunghae-eup, Pohang
*[image: phone-icon]*  010-5636-3680
[image: email-icon]  ***@***.***
Github https://github.com/wanilly

*Commit your way to the *

*LORD, trust in him and he will do this:Psalm 37:5*
[image: facebook icon] <https://m.facebook.com/profile.php> [image:
linkedin icon] <https://www.linkedin.com/in/seok-wan-kim-9b1956206/>

rostest, 2024-11-11:
The scan data is encoded in the LMDscandata telegram, see table 158 on page 97-109 in the [TiM781S telegram listing](https://cdn.sick.com/media/docs/7/27/927/telegram_listing_telegram_listing_ranging_sensors_lms1xx_lms5xx_tim2xx_tim5xx_tim7xx_lms1000_mrs1000_mrs6000_nav310_ld_oem15xx_ld_lrs36xx_lms4000_lrs4000_multiscan100_picoscan100_en_im0045927.pdf).

Note that receiving the LMDscandata telegrams requires proper initialization with the appropriate SOPAS commands. A typical TiM781S startup sequence looks like this (see the telegram listing for details):
```
Sending  : "sRN SCdevicestate"
Receiving: "sRA SCdevicestate \x01"
Sending  : "sMN SetAccessMode 0x03 0x6f 0xd6 0x2c 0x05"
Receiving: "sAN SetAccessMode \x01"
Sending  : "sWN EIHstCola 0x01"
Receiving: "sWA EIHstCola"
Sending  : "sRN FirmwareVersion"
Receiving: "sRA FirmwareVersion \x00\x05\x56\x33\x2e\x31\x33"
Sending  : "sRN SCdevicestate"
Receiving: "sRA SCdevicestate \x01"
Sending  : "sRN ODoprh"
Receiving: "sRA ODoprh \x00\x00\x01\x61"
Sending  : "sRN ODpwrc"
Receiving: "sRA ODpwrc \x00\x00\x00\x17"
Sending  : "sRN LocationName"
Receiving: sRA LocationName \x00\x0b\x6e\x6f\x74\x20\x64\x65\x66\x69\x6e\x65\x64"
Sending  : "sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x0d 0x05 0xff 0xf9 0x22 0x30 0x00 0x22 0x55 0x10"
Receiving: "sWA LMPoutputRange"
Sending  : "sRN LMPoutputRange"
Receiving: "sRA LMPoutputRange \x00\x01\x00\x00\x0d\x05\xff\xf9\x22\x30\x00\x22\x55\x10"
Sending  : "sRN LIDinputstate"
Receiving: "sRA LIDinputstate \x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
Sending  : "sWN LMDscandatacfg 0x01 0x00 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01"
Receiving: "sWA LMDscandatacfg "
Sending  : "sRN LMDscandatacfg"
Receiving: "sRA LMDscandatacfg \x01\x00\x01\x01\x00\x00\x00\x00\x00\x00\x01\x00\x01"
Sending  : "sEN LFErec 0x01"
Receiving: "sEA LFErec \x01"
Sending  : "sEN LIDoutputstate 0x01"
Receiving: "sEA LIDoutputstate \x01"
Sending  : "sEN LIDinputstate 0x01"
Receiving: "sEA LIDinputstate \x01"
Sending  : "sMN LMCstartmeas"
Receiving: "sAN LMCstartmeas \x00"
Sending  : "sMN Run"
Receiving: "sAN Run \x01"
Sending  : "sEN LMDscandata 0x01"
Receiving: "sEA LMDscandata \x01"
```

Please let me recommend again to update the operating system and use sick_scan_xd. Ubuntu 16 and 18 have reached the end of their life cycle a few years ago and are not supported by sick_scan_xd.

closed 2024-12-02 by rostest