SICKAG/sick_scan_xd #415: Sector contains doubled points in the full scan data

Nigel-BMV, 2024-11-05:
We are using the picoscan-150 to check a moving object, however we have noticed that occassionally the message data with the -1 ID has sections that are doubled. This messes with the data of the moving object creating a zone with unsure measurements.
To get the full scan data, I'm using a filter on the SickScanPointCloudMsg segment_idx, where -1 gives me the full scene.

I have recreated the issue with a **static** scene. **103** Scans were taken, of which **8** scans have doubled data.

The data has a length of 333 in normal circumstances and visualised appears like : 
[image]

The doubled data has a length of 393 and visually appears like : 
[image]

The area which contains the doubled data : 
[image]

Debug console messaging, as you can see the message is 333 x 1, and 393 x 1 for the following with the first 60 x 1 doubled : 
```
[Info]: MsgPack/Compact-Exporter:   0 udp packages still in input fifo, 10 messages still in output fifo, current segment index: 9
[Info]: MsgPack/Compact-Exporter: 1740 udp scandata messages received, 2000 messages exported (scan+imu), 0% package lost.
[Info]: MsgPack/Compact-Exporter: max. 2 udp messages buffered, max 21 export messages buffered.
[Info]: MsgPack/Compact-Exporter: 2000 messages exported at 172.114 Hz, mean time: 48.687 milliseconds/messages, stddev time: 25.724, max time: 251.363 milliseconds between udp receive and messages export, histogram=[30,6,8,4,8,7,9,6,8,10,1904]
C++ PointCloudMsgCb: 333 x 1 pointcloud message received
>>>>>> IMAGE CREATION : 09_01_29_126__333
C++ PointCloudMsgCb: 60 x 1 pointcloud message received
C++ PointCloudMsgCb: 60 x 1 pointcloud message received
C++ PointCloudMsgCb: 13 x 1 pointcloud message received
C++ PointCloudMsgCb: 60 x 1 pointcloud message received
C++ PointCloudMsgCb: 60 x 1 pointcloud message received
[Info]: MsgPack/Compact-Exporter:   1 udp packages still in input fifo, 14 messages still in output fifo, current segment index: 1
[Info]: MsgPack/Compact-Exporter: 1752 udp scandata messages received, 2009 messages exported (scan+imu), 0% package lost.
[Info]: MsgPack/Compact-Exporter: max. 2 udp messages buffered, max 21 export messages buffered.
[Info]: MsgPack/Compact-Exporter: 2009 messages exported at 171.341 Hz, mean time: 48.789 milliseconds/messages, stddev time: 25.762, max time: 251.363 milliseconds between udp receive and messages export, histogram=[30,6,8,4,8,7,9,6,8,10,1913]
C++ PointCloudMsgCb: 15 x 1 pointcloud message received
C++ PointCloudMsgCb: 35 x 1 pointcloud message received
C++ PointCloudMsgCb: 21 x 1 pointcloud message received
C++ PointCloudMsgCb: 25 x 1 pointcloud message received
C++ PointCloudMsgCb: 15 x 1 pointcloud message received
C++ PointCloudMsgCb: 29 x 1 pointcloud message received
C++ PointCloudMsgCb: 60 x 1 pointcloud message received
C++ PointCloudMsgCb: 13 x 1 pointcloud message received
C++ PointCloudMsgCb: 333 x 1 pointcloud message received
>>>>>> IMAGE CREATION : 09_01_29_257__333
C++ PointCloudMsgCb: 60 x 1 pointcloud message received
[Info]: MsgPack/Compact-Exporter:   0 udp packages still in input fifo, 20 messages still in output fifo, current segment index: 0
[Info]: MsgPack/Compact-Exporter: 1761 udp scandata messages received, 2024 messages exported (scan+imu), 0% package lost.
[Info]: MsgPack/Compact-Exporter: max. 2 udp messages buffered, max 21 export messages buffered.
[Info]: MsgPack/Compact-Exporter: 2024 messages exported at 170.709 Hz, mean time: 48.911 milliseconds/messages, stddev time: 25.760, max time: 251.363 milliseconds between udp receive and messages export, histogram=[30,6,8,4,8,7,9,6,8,10,1928]
C++ PointCloudMsgCb: 60 x 1 pointcloud message received
C++ PointCloudMsgCb: 60 x 1 pointcloud message received
C++ PointCloudMsgCb: 15 x 1 pointcloud message received
C++ PointCloudMsgCb: 35 x 1 pointcloud message received
C++ PointCloudMsgCb: 21 x 1 pointcloud message received
C++ PointCloudMsgCb: 25 x 1 pointcloud message received
C++ PointCloudMsgCb: 15 x 1 pointcloud message received
C++ PointCloudMsgCb: 29 x 1 pointcloud message received
C++ PointCloudMsgCb: 60 x 1 pointcloud message received
C++ PointCloudMsgCb: 13 x 1 pointcloud message received
C++ PointCloudMsgCb: 393 x 1 pointcloud message received
>>>>>> IMAGE CREATION : 09_01_29_406__393
C++ PointCloudMsgCb: 60 x 1 pointcloud message received
[Info]: MsgPack/Compact-Exporter:   0 udp packages still in input fifo, 20 messages still in output fifo, current segment index: 0
[Info]: MsgPack/Compact-Exporter: 1781 udp scandata messages received, 2043 messages exported (scan+imu), 0% package lost.
[Info]: MsgPack/Compact-Exporter: max. 2 udp messages buffered, max 21 export messages buffered.
[Info]: MsgPack/Compact-Exporter: 2043 messages exported at 170.024 Hz, mean time: 49.124 milliseconds/messages, stddev time: 25.821, max time: 251.363 milliseconds between udp receive and messages export, histogram=[30,6,8,4,8,7,9,6,8,10,1947]

```

Normal scan field data (X/Y/Z/Intensity)

```
-1.132629;-1.019671;0.000000;0.000000;
[... 709 lines cut ...]
-0.547199;0.687905;0.000000;0.000000;
-0.555003;0.685490;0.000000;0.000000;
-0.957309;1.161286;0.000000;0.000000;
-0.951226;1.133830;0.000000;0.000000;
-0.952747;1.115510;0.000000;0.000000;
-0.953809;1.097435;0.000000;0.000000;
-0.956826;1.081490;0.000000;0.000000;
-0.957425;1.063532;0.000000;0.000000;
-0.959337;1.046935;0.000000;0.000000;
-0.961031;1.030388;0.000000;0.000000;
-0.961626;1.013353;0.000000;0.000000;
-0.964275;0.998358;0.000000;0.000000;
-0.965845;0.982867;0.000000;0.000000;
-0.969526;0.969361;0.000000;0.000000;
-0.969296;0.952547;0.000000;0.000000;
-0.967589;0.934236;0.000000;0.000000;
-0.966909;0.917592;0.000000;0.000000;
-0.966920;0.901526;0.000000;0.000000;
-0.967291;0.886393;0.000000;0.000000;
-0.968383;0.871805;0.000000;0.000000;
```

Due to the real setup having objects in motion, meaning a variable amount of points; I am unable to filter this specific situation.

**How do I prevent this from occurring, receiving just a singular full scan at all times?**

rostest, 2024-11-05:
Thank you for your feedback. We will be analysing the behaviour in more detail. Are you using the API on Windows or Linux? Are you using the default settings in the launchfile with full angle range and last echo only? If optional filters are enabled, please post the configuration and the launchfile.

Nigel-BMV, 2024-11-05:
> Are you using the API on Windows or Linux

Windows without ROS

> Are you using the default settings in the launchfile with full angle range and last echo only?

The launch file is still in its default state, but I've included it here : 

`<?xml version="1.0"?>
<launch>

    
    
    <arg name="hostname" default="192.168.0.1"/>  
    <arg name="udp_receiver_ip" default=""/>        
    <arg name="nodename" default="sick_picoscan"/>
    <arg name="publish_frame_id" default="world" />                                 
    <arg name="publish_laserscan_segment_topic" default="scan_segment" />           
    <arg name="publish_laserscan_fullframe_topic" default="scan_fullframe" />       
    <arg name="imu_topic" default="imu" />                                          
    <arg name="udp_port" default="2115" />                                          
    <arg name="imu_udp_port" default="7503"/>                                       
    <arg name="add_transform_xyz_rpy" default="0,0,0,0,0,0"/>
    <arg name="scandataformat" default="2"/>                  
    <arg name="performanceprofilenumber" default="-1"/>       
    <arg name="all_segments_min_deg" default="-138.0" />      
    <arg name="all_segments_max_deg" default="+138.0" />      
    <arg name="host_FREchoFilter" default="2" />              
    <arg name="host_LFPangleRangeFilter" default="0 -138.0 +138.0 -90.0 +90.0 1" /> 
    <arg name="host_set_LFPangleRangeFilter" default="False" />                     
    <arg name="host_LFPintervalFilter" default="0 1" />                             
    <arg name="host_set_LFPintervalFilter" default="False" />                       
    <arg name="custom_pointclouds" default="cloud_unstructured_segments cloud_unstructured_fullframe cloud_polar_unstructured_segments cloud_polar_unstructured_fullframe cloud_all_fields_fullframe"/> 
    <arg name="tf_publish_rate" default="10.0" />                     
    <node name="$(arg nodename)" pkg="sick_scan_xd" type="sick_generic_caller" respawn="false" output="screen" required="true">    
        <param name="scanner_type" type="string" value="sick_picoscan"/>

        
        <param name="hostname" type="string" value="$(arg hostname)" />                     
        <param name="udp_receiver_ip" type="string" value="$(arg udp_receiver_ip)" />       

        
        <param name="udp_sender" type="string" value="" />                                  
        <param name="udp_port" type="int" value="$(arg udp_port)" />                        
        <param name="check_udp_receiver_ip" type="bool" value="True" />                     
        <param name="check_udp_receiver_port" type="int" value="2116" />                    
        <param name="all_segments_min_deg" type="double" value="$(arg all_segments_min_deg)" />              
        <param name="all_segments_max_deg" type="double" value="$(arg all_segments_max_deg)" />              
        <param name="publish_frame_id" type="string" value="$(arg publish_frame_id)" />     
        <param name="publish_laserscan_segment_topic" type="string" value="$(arg publish_laserscan_segment_topic)" />           
        <param name="publish_laserscan_fullframe_topic" type="string" value="$(arg publish_laserscan_fullframe_topic)" />       
        <param name="udp_input_fifolength" type="int" value="20" />                         
        <param name="msgpack_output_fifolength" type="int" value="20" />                    
        <param name="verbose_level" type="int" value="1" />                                 
        <param name="measure_timing" type="bool" value="True" />                            
        <param name="export_csv" type="bool" value="False" />                               
        <param name="export_udp_msg" type="bool" value="False" />                           
        <param name="logfolder" type="string" value="" />                                   
        <param name="send_udp_start" type="bool" value="False" />                           
        <param name="send_udp_start_string" type="string" value="magicalActivate" />        
        <param name="udp_timeout_ms" type="int" value="10000" />                            
        <param name="udp_timeout_ms_initial" type="int" value="60000" />                    
        <param name="scandataformat" type="int" value="$(arg scandataformat)" />            
        <param name="performanceprofilenumber" type="int" value="$(arg performanceprofilenumber)"/> 
        
        <param name="imu_enable" type="bool" value="True"/>                                 
        <param name="imu_udp_port" type="int" value="$(arg imu_udp_port)"/>                 
        <param name="imu_latency_microsec" type="int" value="0"/>                           
        <param name="imu_topic" type="string" value="$(arg imu_topic)"/>                    
        
[... 106 lines cut ...]
        <param name="cloud_all_fields_segments" type="string" value="coordinateNotation=3 updateMethod=1 fields=x,y,z,i,range,azimuth,elevation,t,ts,lidar_sec,lidar_nsec,ring,layer,echo,reflector echos=0,1,2 layers=1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16 reflectors=0,1 infringed=0,1 rangeFilter=0,999,0 topic=/cloud_all_fields_segments frameid=world publish=1"/>

        
        <param name="cloud_all_fields_fullframe" type="string" value="coordinateNotation=3 updateMethod=0 fields=x,y,z,i,range,azimuth,elevation,t,ts,lidar_sec,lidar_nsec,ring,layer,echo,reflector echos=0 [... cut ...]
        <param name="tf_base_frame_id" type="string" value="map" />              
        <param name="tf_base_lidar_xyz_rpy" type="string" value="0,0,0,0,0,0" /> 
        <param name="tf_publish_rate" type="double" value="$(arg tf_publish_rate)" />                

        
        <param name="tick_to_timestamp_mode" type="int" value="0"/>

    </node>

</launch>

`

> If optional filters are enabled

There are no optional filters in use, but we are using a distance limiter

Echo Filter = last Echo
Distance filter 150-2000mm

closed 2024-12-02 by rostest