SICKAG/sick_scan_xd #416: 'discarding message because the queue is full'

lnut220303082, 2024-11-06:
如下报错：[rviz2-2] [INFO] [1730895321.500974115] [rviz2]: Message Filter dropping message: frame 'cloud' at time 1730895320.824 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1730895321.564788616] [rviz2]: Message Filter dropping message: frame 'cloud' at time 1730895320.891 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1730895321.660530351] [rviz2]: Message Filter dropping message: frame 'cloud' at time 1730895320.958 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1730895321.725094776] [rviz2]: Message Filter dropping message: frame 'cloud' at time 1730895321.024 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1730895321.789047697] [rviz2]: Message Filter dropping message: frame 'cloud' at time 1730895321.091 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1730895321.852479631] [rviz2]: Message Filter dropping message: frame 'cloud' at time 1730895321.158 for reason 'discarding message because the queue is full'
[rviz2-2] [INFO] [1730895321.917081471] [rviz2]: Message Filter dropping message: frame 'cloud' at time 1730895321.224 for reason 'discarding message because the queue is full'
雷达tim5661通信成功，launch文件启动成功，rviz显示点云出错。不显示扫描后的地图

rostest, 2024-11-06:
Thank you for your feedback. The messages indicate that rviz2 cannot display point clouds fast enough due to rviz or ros performance bottlenecks. This can have several reasons, most likely related to the system (hardware, OS or ROS2). Use `top`, `ps` to check memory and cpu usage. Monitoring tools provided by the operating system can identify resource bottlenecks.

closed 2024-11-06 by rostest

lnut220303082, 2024-11-07:
已解决，谢谢，thank you