SICKAG/sick_scan_xd #417: Creating a Deep Copy of SickScanPointCloudMsg C++ without RoS

Nigel-BMV, 2024-11-07:
Due to a slowdown in the processing of the callback of SickScanPointCloudMsg, I am attempting to push an exact copy of the message to a heap list to process and save all the data when the scanning has finished.

I am trying to use the functionality of the constructor to create a copy of the CONST msg, however this appears to simply corrupt the actual data buffer.

```
class MessageData {
public : 
    std::string fileName;
    SickScanPointCloudMsg* msg;

    MessageData::MessageData(std::string &_file,  SickScanPointCloudMsg* &_msgptr)
    {
        fileName = _file;
        msg = _msgptr;
    };

};

std::list<MessageData*> pointData;

void customizedPointCloudMsgCb(SickScanApiHandle apiHandle, const SickScanPointCloudMsg* msg)
{
    if (msg->segment_idx < 0)
    {
        std::cout << ">>>>>> FULL SEGMENT : " << std::endl;  
        std::string fileName = time_in_HH_MM_SS_MMM() + "__" + std::to_string(msg->width); 
        SickScanPointCloudMsg* CopyMSG = new SickScanPointCloudMsg(*msg); //constructor has const functionality
      
        pointData.push_back(new MessageData(fileName, CopyMSG));
    }      
}
```

The List container post-scan : 
[image]

[image]

The assert failure because all the offsets are -1 : 
[image]

I cannot seem to be able to correctly copy the buffer or anything to do with it in the callback.

How exactly do I create a Deep Copy of the SickScanPointCloudMsg structure? 

Can a Copy constructor be added with a bool overload for a full deep copy to make the functionality easily accessible?

rostest, 2024-11-08:
Thank you for your feedback. The memory for the scan data in the struct SickScanPointCloudMsg is released after the custom callback has been executed. For data processing outside the callback, a deep copy of the scan data is therefore required.

The MessageData class stores only the pointer to the message, which is destroyed after leaving the `customizedPointCloudMsgCb` function. To create a deep copy of the scan data, you can parse the scan points and convert them to a std::vector like this

```
class ScanPointXYZI {
public:
  float x = 0;
  float y = 0;
  float z = 0;
  float i = 0;
};

// Create a deep copy of the scan data by converting a point cloud message to a vector of cartesian points of type ScanPointXYZI
void convertScanPoints(const SickScanPointCloudMsg& msg, std::vector<ScanPointXYZI>& scan_points)
{
  // Get offsets for x, y, z, intensity values
  SickScanPointFieldMsg* msg_fields_buffer = (SickScanPointFieldMsg*)msg.fields.buffer;
  int field_offset_x = -1, field_offset_y = -1, field_offset_z = -1, field_offset_intensity = -1;
  for (int n = 0; n < msg.fields.size; n++)
  {
    if (strcmp(msg_fields_buffer[n].name, "x") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
      field_offset_x = msg_fields_buffer[n].offset;
    else if (strcmp(msg_fields_buffer[n].name, "y") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
      field_offset_y = msg_fields_buffer[n].offset;
    else if (strcmp(msg_fields_buffer[n].name, "z") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
      field_offset_z = msg_fields_buffer[n].offset;
    else if (strcmp(msg_fields_buffer[n].name, "intensity") == 0 && msg_fields_buffer[n].datatype == SICK_SCAN_POINTFIELD_DATATYPE_FLOAT32)
      field_offset_intensity = msg_fields_buffer[n].offset;
  }
  assert(field_offset_x >= 0 && field_offset_y >= 0 && field_offset_z >= 0);
  // Convert all points in the point cloud to ScanPointXYZI
  scan_points.clear();
  scan_points.resize(msg.height * msg.width);
  for (int row_idx = 0, point_idx = 0; row_idx < (int)msg.height; row_idx++)
  {
    for (int col_idx = 0; col_idx < (int)msg.width; col_idx++, point_idx++)
    {
      // Get cartesian point coordinates
      int polar_point_offset = row_idx * msg.row_step + col_idx * msg.point_step;
      scan_points[point_idx].x = *((float*)(msg.data.buffer + polar_point_offset + field_offset_x));
      scan_points[point_idx].y = *((float*)(msg.data.buffer + polar_point_offset + field_offset_y));
      scan_points[point_idx].z = *((float*)(msg.data.buffer + polar_point_offset + field_offset_z));
      if (field_offset_intensity >= 0)
        scan_points[point_idx].i = *((float*)(msg.data.buffer + polar_point_offset + field_offset_intensity));
    }
  }
}

class MessageData {
public:
  std::string fileName;
  std::vector<ScanPointXYZI> scan_points;
  MessageData::MessageData(const std::string& _file, const SickScanPointCloudMsg* msg)
  {
    fileName = _file;
    convertScanPoints(*msg, scan_points);
  };
};

std::list<MessageData> pointData;

void customizedPointCloudMsgCb(SickScanApiHandle apiHandle, const SickScanPointCloudMsg* msg)
{
  if (msg->segment_idx < 0)
  {
    std::string fileName = "data.bin";
    pointData.push_back(MessageData(fileName, msg));
  }
}
```

Note that `std::list<MessageData> pointData` is not thread-safe. To pop the MessageData from container `pointData` separately, it should be protected by a mutex. Feel free to use [fifo.h](https://github.com/SICKAG/sick_scan_xd/blob/develop/include/sick_scansegment_xd/fifo.h) as an example of a FIFO buffer ("first-in-first-out") with protected push and pop functions.

rostest, 2024-11-08:
I see. To make a deep copy of the plain message data, you can copy the memory referenced by the pointer to a std::vector<uint8_t> like this:

```
class PointCloudMsgData
{
public:
  SickScanHeader header;          // message timestamp
  uint32_t height;                // 2D structure of the point cloud. If the cloud is unordered, height is 1
  uint32_t width;                 // and width is the length of the point cloud.
  std::vector<SickScanPointFieldMsg> fields; // Describes the channels and their layout in the binary data blob.
  uint8_t   is_bigendian;         // Is this data bigendian?
  uint32_t  point_step;           // Length of a point in bytes
  uint32_t  row_step;             // Length of a row in bytes
  std::vector<uint8_t> data;      // Actual point data, size is (row_step*height)
  uint8_t is_dense;               // True if there are no invalid points
  int32_t num_echos;              // number of echos
  int32_t segment_idx;            // segment index (or -1 if pointcloud contains data from multiple segments)
  std::string topic;              // ros topic this pointcloud is published
};

class MessageData {
public:
  std::string fileName;
  PointCloudMsgData data;
  MessageData::MessageData(const std::string& _file, const SickScanPointCloudMsg* msg)
  {
    fileName = _file;
    data.header = msg->header;
    data.height = msg->height;
    data.width = msg->width;
    data.fields.reserve(msg->fields.size);
    for (int n = 0; n < msg->fields.size; n++)
      data.fields.push_back(msg->fields.buffer[n]);
    data.is_bigendian = msg->is_bigendian;
    data.point_step = msg->point_step;
    data.row_step = msg->row_step;
    data.data.resize(msg->data.size);
    for (int n = 0; n < msg->data.size; n++)
      data.data[n] = msg->data.buffer[n];
    data.is_dense = msg->is_dense;
    data.num_echos = msg->num_echos;
    data.segment_idx = msg->segment_idx;
    data.topic = msg->topic;
  };
};

std::list<MessageData> pointData;

void customizedPointCloudMsgCb(SickScanApiHandle apiHandle, const SickScanPointCloudMsg* msg)
{
  if (msg->segment_idx < 0)
  {
    std::string fileName = "data.bin";
    pointData.push_back(MessageData(fileName, msg));
  }
}
```

This copies the message data and memory referenced by pointers into an STL container for later use.

closed 2024-12-02 by rostest