SICKAG/sick_scan_xd #418: Features for LRS4000

stffnb, 2024-11-11:
Hello, 
we are using the LRS4000 and are wondering if the following two features are planned for the LRS 4000?
- dirt contamination service similar to https://github.com/SICKAG/sick_scan_xd/blob/6bdc5a38edcefebc0fc4c42583c67cf52ccb7d79/driver/src/sick_scan_services.cpp#L85-L90 (https://github.com/4am-robotics/orga/issues/6780)
- imu-topic similar to https://github.com/SICKAG/sick_scan_xd/commit/f4b3117cfbe8e6f47dc5089612c60e9c986ef13a (https://github.com/4am-robotics/orga/issues/6781)

rostest, 2024-11-12:
Thank you for your feedback. According to the [telegram listing](https://cdn.sick.com/media/docs/7/27/927/telegram_listing_telegram_listing_ranging_sensors_lms1xx_lms5xx_tim2xx_tim5xx_tim7xx_lms1000_mrs1000_mrs6000_nav310_ld_oem15xx_ld_lrs36xx_lms4000_lrs4000_multiscan100_picoscan100_en_im0045927.pdf), SOPAS commands "sEN InertialMeasurementUnit" to activate IMU data and "sRN ContaminationResult" to read the ContaminationResult are not supported for LRS4000 devices.

[image]

We will clarify, whether ContaminationResults and IMU data can be provided for the LRS4000.

rostest, 2024-11-27:
IMU and contamination data are now supported for LRS-4xxx devices. Please use the latest sick_scan_xd update in branch [feature/issue_update_2412xx](https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx) and install the latest LRS4000 firmware 1.7.0.

Contamination results are available using the ROS services GetContaminationData and GetContaminationResult:
```
rosservice call /sick_lrs_4xxx/GetContaminationData "{}" # Returns the status of contamination in order of the different channels. One entry (byte) for each of 12 channels. Order of 12 channels: (1/2/3/4/5/6/7/8/9/10/11/12). Status 0 = CM NONE, 1 = CM WARN, 2 = CM ERROR.
rosservice call /sick_lrs_4xxx/GetContaminationResult "{}" # Returns contamination result true (1) or false (0), supported by LRS-4000 for compatibility
```
See the latest operating instructions for full details.

IMU data are enabled by default and published on topic "/imu". The default configuration (IMU on, topic "/imu", frame id "imu_link") can be overwritten in the launch file or by command line parameters.

stffnb, 2024-12-02:
Thank you for the fast implementation! We will try it out.

closed 2024-12-02 by stffnb