SICKAG/sick_scan_xd #420: Fieldset does not show on startup

akashjinandra, 2024-11-12:
When using the fieldset evaluation topic, it does not show up unless there is a change in something infringing the field:
for instance it may show up grey like this: 
[image]
You can see the laserscan topic live and upating, but the fieldset topic does not publish anything, unless there is a change. After the change, it lights up correctly and starts publishing:
[image]
I think it would be a good idea to make that specific topic: 
/sick_tim_7xx/lferec latched.

rostest, 2024-11-13:
Thank you for your feedback. The behaviour is intended; the infringment of the fields is visualized after the first LFErec telegram is received from the lidar. As long as the infringement is unknown after startup, the fields are marked grey.

We will clarify if it is possible to query and visualize the field infringment directly after startup.

rostest, 2024-11-13:
An update is now available in the branch [feature/issue_update_2412xx](https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx). It advertises the lferec messages with the option latch=true on ROS1 or transient_local=true on ROS2. Please checkout this update and try again.

rostest, 2024-11-28:
The TiM-7xx field evaluation status is now additionally queried and published after starting sick_scan_xd. An update can be found in the [feature/issue_update_2412xx](https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx) branch. 

With this update, the field infringment is published both on startup and after a change. You can see the coloured fieldset in rviz immediately after startup. If rviz is started after sick_scan_xd and therefore the first LFErec or Marker message is missing, you can query the field status with the command `rosservice call /sick_tim_7xx/ColaMsg "{request: 'sRN LFErec'}"`. After that you can see the fields and their infringement in the MarkerArray published in topic `/sick_tim_7xx/marker` and in the LFErec message in topic `/sick_tim_7xx/lferec`.

closed 2024-12-20 by rostest