SICKAG/sick_scan_xd #422: Installation failed: Unable to load cache

SICK-AliciaTee, 2024-11-13:
Hi, I am having issue installing the sick_scan_xd driver Windows version without ROS. I have Visual Studio 2022 v17.8.4, would that be an issue?

Please find log attached.

Thank you. 
[Log20241113.txt](https://github.com/user-attachments/files/17740341/Log20241113.txt)

rostest, 2024-11-14:
Thanks for your feedback. The cmake generator string for Visual Studio should be "Visual Studio 17 2022" (instead of "Visual Studio 17.8.4 2022"). To build with Visual Studio 2022, run the following commands:

```
set _os=x64
set _cmake_string=Visual Studio 17 2022
set _msvc=Visual Studio 2022
set _cmake_build_dir=build
if not exist %_cmake_build_dir% mkdir %_cmake_build_dir%
pushd %_cmake_build_dir%
cmake -DROS_VERSION=0 -G "%_cmake_string%" ..
```
Open sick_scan_xd.sln in Visual Studio 2022 and build.

Depending on your system, you may need to set the Visual Studio environment explicitly by calling "VsDevCmd.bat":
`call "%ProgramFiles(x86)%\Microsoft Visual Studio\2022\Community\Common7\Tools\VsDevCmd.bat" -arch=amd64 -host_arch=amd64`
Change the path to Visual Studio if it is installed in a different folder.

SICK-AliciaTee, 2024-11-16:
Hi ROSTest,

Sorry I tried the method above but encounter the error
CMake Warning:
  Ignoring extra path from command line:

   ".."

CMake Error: The source directory "C:/Users/teebo" does not appear to contain CMakeLists.txt.
Specify --help for usage,
[Log20241115.txt](https://github.com/user-attachments/files/17782693/Log20241115.txt)
 or press the help button on the CMake GUI

Please see log attached. 

Also I tried looking VSDevCmd.bat, but cannot find the directory VisualStudio2022
(See photo)
[image]

Thank you. 

Best Regard, 
Alicia Tee

rostest, 2024-11-18:
Thanks for following up. You have to clone the sick_scan_xd repository first into a folder (`sick_scan_xd` by default) and then execute these commands in that folder (i.e. in folder `sick_scan_xd`). Here is the complete command sequence to clone and build sick_scan_xd with Visual Studio 2022:
```
mkdir ws_sick_scan_xd\src
cd ws_sick_scan_xd\src
git clone https://github.com/SICKAG/sick_scan_xd.git
cd sick_scan_xd
set _os=x64
set _cmake_string=Visual Studio 17 2022
set _msvc=Visual Studio 2022
set _cmake_build_dir=build
mkdir %_cmake_build_dir%
pushd %_cmake_build_dir%
cmake -DROS_VERSION=0 -G "%_cmake_string%" ..
REM Open sick_scan_xd.sln in Visual Studio 2022 or run cmake --build .
REM cmake --build . --clean-first --config Debug
REM cmake --build . --clean-first --config Release
popd
```
The workspace folder `ws_sick_scan_xd` is just an example, feel free to rename it. 
Open file `ws_sick_scan_xd\sick_scan_xd\build\sick_scan_xd.sln` in Visual Studio 2022 and build.
Depending on your system, you may need to set the Visual Studio environment explicitly by calling "VsDevCmd.bat":
`call "%ProgramFiles(x86)%\Microsoft Visual Studio\2022\Community\Common7\Tools\VsDevCmd.bat" -arch=amd64 -host_arch=amd64`
Change the path to Visual Studio if it is installed in a different folder.

closed 2024-12-02 by rostest