SICKAG/sick_scan_xd #424: lidar/imu data from Sick PicoScan 150 Core does not recover from a power cycle of the sensor or disconnect and reconnect the cable

mohamedbenftima, 2024-11-14:
Hello,

For my use case, I frequently need to perform a power cycle on the LiDAR or unplug and then replug the Ethernet cable connecting the LiDAR to the PC. However, after doing this, the LiDAR and IMU data do not always reinitialize, and the corresponding topics remain empty. I am using the sick_picoscan.launch file, with the parameter <param name="udp_timeout_ms" value="2000" />.

I conducted two tests and collected the corresponding logs:

    Test 1(plugged): I unplugged and replugged the Ethernet cable, but the LiDAR data did not come back.
    Test 2(lidar_power_cycle.txt): I performed a power cycle, but again, the LiDAR data was not recovered.

Please note that I am using ROS2 and version 3.6.0 of sick_scan_xd.

[lidar_power_cycle.txt](https://github.com/user-attachments/files/17749328/lidar_power_cycle.txt)
[plugged.txt](https://github.com/user-attachments/files/17749330/plugged.txt)

Best regards,
Mohamed

gschwsiSICKAG, 2024-11-19:
Hi Mohamed, thanks for the request. We will have a look at it.

rostest, 2024-11-22:
Thank you for your feedback. Using the latest picoScan firmware version 1.5.0, the driver successfully recovers after both unplugging and replugging the ethernet cable and power cycling. The maximum timeout for the recovery is 60 seconds by default and can be configured using the `udp_timeout_ms_initial` parameter in the launch file. 
Please update the picoScan firmware to the latest version 1.5.0.

mohamedbenftima, 2024-11-25:
Hello,

Thank you for your feedback.  

Here is our updated launch file, where we added the following parameter:  
`<param name="udp_timeout_ms_initial" value="10000" />`  
[sick_picoscan.launch.txt](https://github.com/user-attachments/files/17900279/sick_picoscan.launch.txt)

The results are not entirely conclusive, but here are the trends observed, regardless of the tested versions (1.5.0):  

- Quick OFF/ON (a few seconds): the system tends to recover correctly.  
- Slow OFF/ON (tens of seconds or more): nothing happens, but after 1 to 2 minutes, the driver resets and starts working again.  

It seems that the driver tends to get stuck during the TCP connection setup with the PC if the LiDAR is not powered on at that moment. Once stuck in this phase, nothing happens for 1 to 2 minutes until it resets.  

However, during afresh star (LiDAR powered on before launching the driver), there are never any issues.  

Thus:  
- If the LiDAR is powered on before the driver is launched, it always seems to work correctly.  
- If the driver is launched before the LiDAR is powered on, the behavior becomes unpredictable. Sometimes there’s a delay of 1 to 2 minutes (2 minutes 20 seconds at worst in my tests) before receiving IMU and LiDAR measurements.  

For reference, my firmware version before the update was 1.4.1. 

Diagnostic Logs : 
[SN24360016_1.5.0_1_1_1970-1-28-26AM_DiagnosticDump.zip](https://github.com/user-attachments/files/17900423/SN24360016_1.5.0_1_1_1970-1-28-26AM_DiagnosticDump.zip)
[SN24360016_1.4.1_1_1_1970-1-11-54AM_DiagnosticDump_2024_1125_00h01.zip](https://github.com/user-attachments/files/17900424/SN24360016_1.4.1_1_1_1970-1-11-54AM_DiagnosticDump_2024_1125_00h01.zip)
 

Best regards

rostest, 2024-11-25:
Thanks for the files and the follow-up. Your launchfile shows some modifications (parameters imu_topic and tick_to_timestamp_mode are missing, some type qualifiers like type="int" or type="bool" are removed, parameters sopas_tcp_port and laserscan_layer_filter have additional `''` characters in their values). To avoid side effects, please use the default launch file [sick_picoscan.launch](https://github.com/SICKAG/sick_scan_xd/blob/develop/launch/sick_picoscan.launch) from the github repository.
The zip files are encrypted. Can you send us the password?

gschwsiSICKAG, 2024-11-25:
Thanks for the files. We also have the password.

Please also activate the UDP output: 

[image]

Actually, this is enabled by the driver itself during the launch of the driver. If you activate this before (and save them permanently) this could help during the start-up of the picoScan and the driver.

mohamedbenftima, 2024-11-26:
In order to solve our issue, we had to modify the way the TCP connection is established by adding a timeout.
We have set a delay of 2 seconds.  If the timeout is reached, the driver is then resetting and it retries to establish the TCP connection.

To do so, we have added the following piece of code in file "sick_scan_xd/driver/src/tcp/tcp.cpp" at line 170:

// Setting a timeout on the TCP connection

struct timeval timeout;

timeout.tv_sec = 2; // Set timeout seconds

timeout.tv_usec = 0; // Set timeout microseconds

if (setsockopt(m_connectionSocket, SOL_SOCKET, SO_SNDTIMEO, (const char*)&timeout, sizeof(timeout)) < 0) {

ROS_ERROR("Tcp::open: setsockopt failed to set timeout, aborting.");

close();

return false;

}

Do you see any side effects that could lead to issues arising somewhere else in the driver because of this modification?
In our tests it is working perfectly fine but we would like you to validate it.

For your information, we are using the [sick_picoscan.launch.py](http://sick_picoscan.launch.py/) to launch the driver.

Before this modification, we got the following behavior:

Fresh start
...
(Driver running OK)
....
Power cycle OFF on the LiDAR
...
Driver reset and tries to establish the TCP connection:
[sick_generic_caller-1] [INFO] [1732615219.045835717] [sick_scan_xd]: sick_scan_xd: Tcp::open: connecting to [192.168.0.1:2111](http://192.168.0.1:2111/) ...
...
After a few seconds, power cycle ON on the LiDAR
...
Nothing happens
...
Driver timeout after more than 2 minutes:
[sick_generic_caller-1] [ERROR] [1732615349.462896725] [sick_scan_xd]: Tcp::open: Failed to open TCP connection to [192.168.0.1:2111](http://192.168.0.1:2111/), aborting.
...
Driver reset and then runs OK

Now we do not have this huge delay anymore thanks to the 2secs timeout.

rostest, 2024-11-26:
Thank you for your feedback! We will test and integrate your solution.

We found an alternative by reducing the TCP timeout configured by the `tcp_syn_retries` system parameter:
Edit the file /etc/sysctl.conf by `sudo gedit /etc/sysctl.conf` and add the line
```
net.ipv4.tcp_syn_retries = 2
```
Reinit sysctl with `sudo sysctl -p` and check the setting with `cat /proc/sys/net/ipv4/tcp_syn_retries`. It should show a value of 2.

With this TCP configuration, re-initialisation was successful within a few seconds after the picoScan LED turning green after a power cycle.

rostest, 2024-11-27:
The latest sick_scan_xd update in branch [feature/issue_update_2412xx](https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx) now includes your modification to solve the issue. Thanks again for your input!