SICKAG/sick_scan_xd #425: std: runtime error for Negative time

anath93, 2024-11-15:
The master branch has a issue, where runtime throws error and crashes the Node with std:: negative time error cannot be stored, It looks to me software time sync issue, does not happen in the previous release so ported back for now. Platform is arm64 Jetson Orin, ros2 iron.

Config:
[image]

rostest, 2024-11-18:
Thanks for your feedback. The configuration looks unusual:

* It looks like the configuration does not provide a launchfile (sick_tim_7xx.launch for a TiM-7xx). Note that the launch file provides the default configuration for a lidar family; a missing launch file may cause unexpected behaviour. We highly recommend to use the provided launch file sick_tim_7xx.launch and to overwrite parameter if required. Normally this is just the hostname (the lidars ip address), ros topic and frame id. Use [sick_tim_7xx.launch.py](https://github.com/SICKAG/sick_scan_xd/blob/develop/launch/sick_tim_7xx.launch.py) for launching sick_scan_xd directly or take it as an example for function generate_launch_description. A configuration using sick_tim_7xx.launch can be [sick_tim_7xx.launch.py](https://github.com/SICKAG/sick_scan_xd/blob/develop/launch/sick_tim_7xx.launch.py) or may e.g. look like this:
   ```
   def generate_launch_description():
       ld = LaunchDescription()
       sick_scan_pkg_prefix = get_package_share_directory('sick_scan_xd')
       launch_file_path = os.path.join(sick_scan_pkg_prefix, 'launch/sick_tim_7xx.launch')
       node_arguments=[launch_file_path]
       node_arguments.append("hostname:=192.168.0.1")
       node_arguments.append("laserscan_topic:=scan_obstacle")
       node_arguments.append(f"frame_id:={robot_name}/nav2_obstacle")
       node = Node(
           package='sick_scan_xd',
           executable='sick_generic_caller',
           output='screen',
           arguments=node_arguments
       )
       ld.add_action(node)
       return ld
   ```

* A time_offset = -0.47 seconds is configured. A negative time_offset might cause negative timestamps. A time_offset of -0.47 suggests a mean delay of 470 milliseconds for tcp packets, which is unusual. We recommend to use default value 0 unless you are very sure about network latencies.

Please run the following steps:
* Run sick_scan_xd as described in [README.md](https://github.com/SICKAG/sick_scan_xd?tab=readme-ov-file#running-the-driver) using `ros2 launch sick_scan_xd sick_tim_7xx.launch.py hostname:=192.168.0.1 laserscan_topic:=scan_obstacle frame_id:=nav2_obstacle`
* If this works as expected, use the generate_launch_description above.
* If this works as expected, adapt the parameters in sick_tim_7xx.launch as needed.

anath93, 2024-11-18:
@rostest Thank you, I got it to work. It was my dds settings which was causing latency in this case which was cyclone and need adjustments on the kernel buffer side.

anath93, 2024-11-18:
@rostest, I did what you said and checked dds and kernel buffers additionally. There is a issue somewhere for the bloom release iron on arm64 platform as prior versions did not do that and my launch file has been same. I also removed the time offset but result is same.

rostest, 2024-11-21:
Thanks for following up. We cannot reproduce this error, but have additionally secured the timestamp against negative values. Please use the update in branch [feature/issue_update_2412xx](https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx), rebuild and retry. Please try at first with the default configuration using `ros2 launch sick_scan_xd sick_tim_7xx.launch.py hostname:=192.168.0.1 laserscan_topic:=scan_obstacle frame_id:=nav2_obstacle`.

rostest, 2024-12-03:
The issues #428 and #425 are possibly related.

anath93, 2024-12-03:
@rostest The fix does not still solve the issue, have ported to previous build for now.