SICKAG/sick_scan_xd #426: How to run multiple scanners using cpp, without ROS?

vvranjek, 2024-11-20:
I am deep into a cpp (linux, non-ROS) project where I implemented the sick_scan_xd shared library and use it to read and process data from the scanner. Now the requirements are that we should be able to use multipe scanners at once. After trying to make it work for several days I am starting realize that it is most likely not possible without major modifications to this source code. Am I wrong? Was the software designed to handle multiple scanners at once? If so, what is right approach? Thanks for the help!

rostest, 2024-11-21:
Thank you for your feedback. Your assumption is correct. Currently the sick_scan_xd API does not support single or multi-threaded use of 2 or more lidars in one process, as the sick_scan_xd library is not guaranteed to be thread-safe. To run multiple lidars simultaneously, we recommend using ROS or running sick_scan_xd in multiple and separate processes, so that each process serves one sensor. See [starting multiple nodes](https://github.com/SICKAG/sick_scan_xd?tab=readme-ov-file#start-multiple-nodes) for details.

gschwsiSICKAG, 2024-11-21:
@rostest, thanks for your feedback. @vvranjek: Can you tell us what device you are working with?

vvranjek, 2024-11-22:
I am working with LMS511. I partially solved the issue by building a slightly modified cpp example and running it as a subprocess using popen(). This works with multiple scanner but I haven't found a good way to move the message from the subprocess to the main program. What would be the most efficient way to get SickScanPointCloudMsgType from the suprocess? I am also thinking If I should be dealing with the raw message at this point and parsing it in the main program? At the moment I am just subscribing to   `SickScanApiDeregisterCartesianPointCloudMsg(apiHandle, &customizedPointCloudMsgCb);` however printing and reading `customizedPointCloudMsgCb` doesn't make sense here, thus would like to be dealing with a serialized data. Raw data? Please advise. Thanks

rostest, 2024-11-22:
Thanks for your reply. Running each lidar with sick_scan_xd in its own (sub-)process is a perfect start. To pass each of the point clouds to the main process, you need to
* serialise the callback message of type SickScanPointCloudMsg into a binary buffer,
* transfer the buffer using inter-process communication (IPC),
* receive, convert and handle the point cloud data in the main process.

See [creating a deep copy of SickScanPointCloudMsg](https://github.com/SICKAG/sick_scan_xd/issues/417#issuecomment-2464302392) for an example of serialising the callback message. 
There are several options for the inter-process communication such as IPC frameworks, mqtt, openmp, shared memory, pipes or data transfer via tcp sockets.

closed 2024-12-20 by rostest