SICKAG/sick_scan_xd #427: I have a question, about the warning text that comes up when running the tim781 driver.

m20ano, 2024-11-21:
Hello,

The environment is ros2,humble

My question is, can I ignore the warning text that comes up when I run the tim781 driver? Also, what is the solution to this warning statement?

input

ros2 run sick_scan_xd sick_generic_caller ./src/sick_scan_xd/launch/sick_tim_7xx.launch 

The warning statement that comes up

[INFO] [1732171496.635693137] [sick_scan_xd]: 1 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1732171496.698718534] [sick_scan_xd]: 2 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1732171496.766455423] [sick_scan_xd]: 3 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1732171496.833415781] [sick_scan_xd]: 4 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1732171496.899821259] [sick_scan_xd]: 5 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1732171496.966854247] [sick_scan_xd]: Software PLL is ready and locked now!
[WARN] [1732171497.033300821] [sick_scan_xd]: 7 of 7 packets dropped (100.0 perc.), maxAbsDeltaTime=0.000
[WARN] [1732171497.033372803] [sick_scan_xd]: More packages than expected were dropped!!
Check the network connection.
Check if the system time has been changed in a leap.
If the problems can persist, disable the software PLL with the option sw_pll_only_publish=False  !

I saw your next issue.
「https://github.com/SICKAG/sick_scan_xd/issues/232」

I tried checking the system time and re-installing the package, but could not solve the problem.
I tried making changes to the OPTIONS but to no avail.
Also, is the input for sw_pll_only_publish=False correct for sick_tim_7xx_launch?

I would appreciate it if you could tell me what operation to perform.

rostest, 2024-11-21:
Thank you for your feedback. The warning occurs when there is a significant delay between
* the system time stamp when a lidar telegram is received, and 
* the expected time stamp calculated from lidar ticks by the [Software PLL](https://github.com/SICKAG/sick_scan_xd/tree/master?tab=readme-ov-file#timestamps-and-synchronization-software-pll).

Such a delay can be caused by high network traffic and can lead to inaccurate time stamps. You can tolerate network latency and these messages if they occur occasionally. However, the warning "7 of 7 packets dropped (100.0 perc.)" indicates a problem with timestamps and network latency.

You can disable the software PLL and timestamp warnings with the `sw_pll_only_publish:=false` option; in this case the timestamps are the system timestamps when the scan data is received, not the lidar measurement timestamps. Use this option (i.e. sw_pll_only_publish:=false) if you do not need the precise measurement timestamps. Otherwise, check the network latencies if the problem occurs more often. Make sure you are using the latest sick_scan_xd release and firmware version.

m20ano, 2024-11-22:
Does doing sw_pll_only_publish:=false have a significant impact on SLAM?
I was able to disable the warning if there is no impact and I would like to run it as is.

rostest, 2024-11-22:
Using option `sw_pll_only_publish:=false`, the measurement timestamps are replaced by the time of receiving the scan data from the lidar. If the network latencies are small, the impact on SLAM should be small. 

The following examples may show the effect of disabling the software pll with the sw_pll_only_publish:=false option:
* The network has a jitter of 1 millisecond. A robot or carrier moves at 2 m/s (7 km/h) or 2 millimetres within the time jitter. An inaccuracy of 2 millimetres is acceptable under normal conditions.
* The network has a high jitter of 100 milliseconds. The carrier moves at 14 m/s (50 km/h), i.e. 1.4 metres within the time jitter. This is probably too much for navigation.

I suggest the following steps:
* Use sw_pll_only_publish:=false and check if the whole system including SLAM works as expected. 
* Check the network latencies. If you are experiencing significant delays, these delays may be affecting not only SLAM, but possibly other system components as well.

m20ano, 2024-11-22:
Yes, sir. Thank you very much. I will try to implement it.

closed 2024-11-25 by m20ano