SICKAG/sick_scan_xd #428: Getting runtime error with raspberry pi 5

purpleLightningG, 2024-11-26:
Hi, I am using a MRS6124R lidar with a pi 5. When I try to run the launch file(sick_mrs_6xxx.launch) I get the error:

terminate called after throwing an instance of 'std::runtime_error'
       what(): Time is out of dual 32 bit range

All my docker, container, os are in 64 bit. I have tried multiple ways to fixed these issues but nothing is solving it. Any idea how can I get rid of this issue? Below is the overall message.

roslaunch sick_scan_xd sick_mrs_6xxx.launch hostname:=192.168.0.3

... logging to /root/.ros/log/cc42a520-ac2e-11ef-b8b4-0242ac110002/roslaunch-a2a982823167-2928.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://a2a982823167:42339/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.17.0
 * /sick_mrs_6xxx/add_transform_check_dynamic_updates: False
 * /sick_mrs_6xxx/add_transform_xyz_rpy: 0,0,0,0,0,0
 * /sick_mrs_6xxx/client_authorization_pw: F4724744
 * /sick_mrs_6xxx/cloud_topic: cloud
 * /sick_mrs_6xxx/filter_echos: 2
 * /sick_mrs_6xxx/frame_id: cloud
 * /sick_mrs_6xxx/hostname: 192.168.0.3
 * /sick_mrs_6xxx/imu_topic: imu
 * /sick_mrs_6xxx/laserscan_topic: scan
 * /sick_mrs_6xxx/max_ang: 1.047
 * /sick_mrs_6xxx/message_monitoring_enabled: True
 * /sick_mrs_6xxx/min_ang: -1.047
 * /sick_mrs_6xxx/min_intensity: 0.0
 * /sick_mrs_6xxx/port: 2112
 * /sick_mrs_6xxx/range_filter_handling: 0
 * /sick_mrs_6xxx/range_max: 250.0
 * /sick_mrs_6xxx/range_min: 0.1
 * /sick_mrs_6xxx/read_timeout_millisec_default: 5000
 * /sick_mrs_6xxx/read_timeout_millisec_kill_node: 150000
 * /sick_mrs_6xxx/read_timeout_millisec_startup: 120000
 * /sick_mrs_6xxx/ros_qos: -1
 * /sick_mrs_6xxx/scandatacfg_timingflag: -1
 * /sick_mrs_6xxx/scanner_type: sick_mrs_6xxx
 * /sick_mrs_6xxx/start_services: True
 * /sick_mrs_6xxx/sw_pll_only_publish: False
 * /sick_mrs_6xxx/tf_base_frame_id: map
 * /sick_mrs_6xxx/tf_base_lidar_xyz_rpy: 0,0,0,0,0,0
 * /sick_mrs_6xxx/tf_publish_rate: 10.0
 * /sick_mrs_6xxx/tick_to_timestamp_mode: 0
 * /sick_mrs_6xxx/timelimit: 5
 * /sick_mrs_6xxx/use_binary_protocol: True
 * /sick_mrs_6xxx/use_generation_timestamp: True

NODES
  /
    sick_mrs_6xxx (sick_scan_xd/sick_generic_caller)

ROS_MASTER_URI=http://localhost:11311/

process[sick_mrs_6xxx-1]: started with pid [2942]
[ INFO] [1732650682.379896226]: sick_generic_caller V. 3.6.0
[ INFO] [1732650682.382449346]: Program argument 1: /root/catkin_ws/devel/lib/sick_scan_xd/sick_generic_caller
[ INFO] [1732650682.382534420]: Program argument 2: __log:=/root/.ros/log/cc42a520-ac2e-11ef-b8b4-0242ac110002/sick_mrs_6xxx-1.log
[ INFO] [1732650682.382553031]: Program argument 3: __name:=sick_mrs_6xxx
[ INFO] [1732650682.382570791]: ==========================================
[ INFO] [1732650682.394577348]: SickTransformPublisher: broadcasting TF messages parent_frame_id="map", child_frame_id="cloud", (x,y,z,roll,pitch,yaw)=(0.0,0.0,0.0,0.0,0.0,0.0), rate=10.0
[ INFO] [1732650682.397012911]: Range filter configuration: range_min=0.1, range_max=250, range_filter_handling=0
[ INFO] [1732650682.397931358]: Found sopas_protocol_type param overwriting default protocol:
[... 60 lines cut ...]
[ INFO] [1732650685.337266093]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[ INFO] [1732650685.337344723]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[ INFO] [1732650685.338533227]: Receiving: <STX>sAN Run \x01<ETX>
[ INFO] [1732650685.338620542]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1732650685.339539453]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ INFO] [1732650685.341005384]: SickScanServices: service "/sick_mrs_6xxx/ColaMsg" created ("/sick_mrs_6xxx/ColaMsg")
[ INFO] [1732650685.341547793]: SickScanServices: service "/sick_mrs_6xxx/ECRChangeArr" created ("/sick_mrs_6xxx/ECRChangeArr")
[ INFO] [1732650685.342084795]: SickScanServices: service "/sick_mrs_6xxx/LIDoutputstate" created ("/sick_mrs_6xxx/LIDoutputstate")
[ INFO] [1732650685.342623334]: SickScanServices: service "/sick_mrs_6xxx/SCdevicestate" created ("/sick_mrs_6xxx/SCdevicestate")
[ INFO] [1732650685.343119428]: SickScanServices: service "/sick_mrs_6xxx/SCreboot" created ("/sick_mrs_6xxx/SCreboot")
[ INFO] [1732650685.343605319]: SickScanServices: service "/sick_mrs_6xxx/SCsoftreset" created ("/sick_mrs_6xxx/SCsoftreset")
[ INFO] [1732650685.344067820]: SickScanServices: service "/sick_mrs_6xxx/SickScanExit" created ("/sick_mrs_6xxx/SickScanExit")
[ INFO] [1732650685.344098894]: SickScanServices: ros services initialized
[ INFO] [1732650685.344116654]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"
terminate called after throwing an instance of 'std::runtime_error'
  what():  Time is out of dual 32-bit range
================================================================================REQUIRED process [sick_mrs_6xxx-1] has died!
process has died [pid 2942, exit code -6, cmd /root/catkin_ws/devel/lib/sick_scan_xd/sick_generic_caller __name:=sick_mrs_6xxx __log:=/root/.ros/log/cc42a520-ac2e-11ef-b8b4-0242ac110002/sick_mrs_6xxx-1.log].
log file: /root/.ros/log/cc42a520-ac2e-11ef-b8b4-0242ac110002/sick_mrs_6xxx-1*.log
Initiating shutdown!
================================================================================
[sick_mrs_6xxx-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

rostest, 2024-11-27:
Thank you for your feedback. What operating system are you using?

We are having trouble reproducing this error. Can you run sick_scan_xd in gdb and see the details of this exception? 
Please rebuild sick_scan_xd without optimisation and with full debugging information (cmake flag `-DO=0`). To run sick_scan_xd in gdb, add the option `launch-prefix="gdb -ex run --args"` to the launchfile and restart: 
`<node name="$(arg nodename)" pkg="sick_scan_xd" type="sick_generic_caller" launch-prefix="gdb -ex run --args" respawn="false" output="screen" required="true">`.
If you get the runtime exception, please watch the stack and variables with the [gdb commands](https://github.com/reveng007/GDB-Cheat-Sheet#stack-backtrace) such as bt, up, down, info, print and post a copy or screenshots.

purpleLightningG, 2024-12-02:
Hi, thanks for your reply. 

I am using raspbian 64 bit OS. After posting my issue above I also tried to rebuild using this command (catkin_make -DCMAKE_CXX_FLAGS="-DLDMRS=0 -DRASPBERRY=1") as suggested in this thread(https://github.com/SICKAG/sick_scan_xd/issues/303) but got the same error.

I then proceeded with your above suggestions and made the following changes to run the file:

1) used this command to rebuild sick scan without optimisation
catkin_make -DCMAKE_BUILD_TYPE=Debug -DO=0

2) made these changes in the launch file (a screenshot is attached as well for the launch file)
**<node name="$(arg nodename)" pkg="sick_scan_xd" type="sick_generic_caller" launch-prefix="gdb -ex run --args" respawn="false" output="screen" required="true">.**

3) ran this command
roslaunch sick_scan_xd sick_mrs_6xxx.launch hostname:=192.168.0.3

got this error:
RLException: Roslaunch got a 'No such file or directory' error while attempting to run:

gdb -ex run --args /root/catkin_ws/devel/lib/sick_scan_xd/sick_generic_caller __name:=sick_mrs_6xxx __log:=/root/.ros/log/89266ccc-b0cf-11ef-aba9-0242ac110002/sick_mrs_6xxx-1.log

Please make sure that all the executables in this command exist and have
executable permission. This is often caused by a bad launch-prefix.
The traceback for the exception was written to the log file
[sick_mrs_6xxx-1] killing on exit

If I use the rosrun command to directly run the launch file I get the same runtime error
rosrun sick_scan_xd sick_generic_caller __name:=sick_mrs_6xxx

[image]

purpleLightningG, 2024-12-02:
Here is the image of the error

[image]

purpleLightningG, 2024-12-02:
Hi, a bit of update here, I realized I never installed gdb. After installing it and running the launch file again I getting some additional lines (image attatched) after the 32 bit runtime error and this time I am able to get some data from the lidar finally. But I am getting the data only for a few seconds before the launch process terminates again.

[image]

rostest, 2024-12-03:
Thanks for the update! At the `(gdb)` prompt, insert the `bt` command to print the stack backtrace. Then walk through the stack frames with the `up` command, print the values of local variables with `info locals` and `info args`, and repeat until the end of the stack backtrace is reached. Please post the gdb output (screenshot or logfile). If you find a suspicious time value, print the value with `display` or `show`. See the [GDB cheat sheet](https://github.com/reveng007/GDB-Cheat-Sheet#stack-backtrace) for details.

Just to be on the safe side: According to reports in [moveit/issues/1062](https://github.com/moveit/moveit/issues/1062) and [robotics.stackexchange.com/questions/88896](https://robotics.stackexchange.com/questions/88896/time-is-out-of-dual-32-bit-range), a "time is out of dual 32 bit range" exception might be caused by an incomplete update of ros packages. Make sure that `sudo apt-get upgrade` has installed the latest releases.

rostest, 2024-12-03:
The issues #428 and #425 are possibly related.

rostest, 2024-12-04:
Thank you for your follow-up. The backtrace was really helpful. Depending on the system and its latencies, timestamps calculated from lidar ticks could be out of range. 
There is an update in the branch [feature/issue_update_2412xx](https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx) with corrected timestamp handling. Please check out this update with `git clone -b feature/issue_update_2412xx https://github.com/SICKAG/sick_scan_xd.git`, rebuild and try again.

purpleLightningG, 2024-12-04:
Thank you very much. Now I am finally able to get the data continuously!

closed 2024-12-04 by purpleLightningG