SICKAG/sick_scan_xd #435: Don't receive IMU data with multiscan100

vanbacvb, 2024-12-12:
i am not getting data from IMU. i am using ros2 humble.

   
    
    <arg name="hostname" default="107.125.137.108"/>                                    
    <arg name="udp_receiver_ip" default="107.125.137.123"/>                                        
    <arg name="nodename" default="multiScan"/>
    <arg name="publish_frame_id" default="world" />                                 
    <arg name="publish_laserscan_segment_topic" default="scan_segment" />           
    <arg name="publish_laserscan_fullframe_topic" default="scan_fullframe" />       
    <arg name="imu_topic" default="/imu" />                                          
    <arg name="udp_port" default="2115" />                                          
    <arg name="imu_udp_port" default="7503"/>                                       
    <arg name="add_transform_xyz_rpy" default="0,0,0,0,0,0"/>
    <arg name="add_transform_check_dynamic_updates" default="false"/>               
    <arg name="laserscan_layer_filter" default="0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0" /> 
    <arg name="scandataformat" default="2"/>                                        
    <arg name="host_LFPangleRangeFilter" default="0 -180.0 +179.0 -90.0 +90.0 1" /> 
    <arg name="host_set_LFPangleRangeFilter" default="False" />                     
    <arg name="host_LFPlayerFilter" default="0 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1" />  
    <arg name="host_set_LFPlayerFilter" default="False" />                          
    <arg name="host_LFPintervalFilter" default="0 1" />                             
    <arg name="host_set_LFPintervalFilter" default="False" />                       
    <arg name="custom_pointclouds" default="cloud_unstructured_segments cloud_unstructured_fullframe cloud_polar_unstructured_segments cloud_polar_unstructured_fullframe cloud_all_fields_fullframe"/> 
    <arg name="tf_publish_rate" default="10.0" />                                   
    <node name="$(arg nodename)" pkg="sick_scan_xd" type="sick_generic_caller" respawn="false" output="screen" required="true">    
        <param name="scanner_type" type="string" value="sick_multiscan"/>

        
        <param name="hostname" type="string" value="$(arg hostname)" />                     
        <param name="udp_receiver_ip" type="string" value="$(arg udp_receiver_ip)" />       

        
        <param name="udp_sender" type="string" value="" />                                  
        <param name="udp_port" type="int" value="$(arg udp_port)" />                        
        <param name="check_udp_receiver_ip" type="bool" value="True" />                     
        <param name="check_udp_receiver_port" type="int" value="2116" />                    
        <param name="segment_count" type="int" value="12" />                                
        <param name="publish_frame_id" type="string" value="$(arg publish_frame_id)" />     
        <param name="publish_laserscan_segment_topic" type="string" value="$(arg publish_laserscan_segment_topic)" />           
        <param name="publish_laserscan_fullframe_topic" type="string" value="$(arg publish_laserscan_fullframe_topic)" />       
        <param name="udp_input_fifolength" type="int" value="20" />                         
        <param name="msgpack_output_fifolength" type="int" value="20" />                    
        <param name="verbose_level" type="int" value="1" />                                 
        <param name="measure_timing" type="bool" value="True" />                            
        <param name="export_csv" type="bool" value="False" />                               
        <param name="export_udp_msg" type="bool" value="False" />                           
        <param name="logfolder" type="string" value="" />                                   
        <param name="udp_timeout_ms" type="int" value="10000" />                            
        <param name="udp_timeout_ms_initial" type="int" value="60000" />                    
        <param name="scandataformat" type="int" value="$(arg scandataformat)" />            
        <param name="imu_enable" type="bool" value="True"/>                                 
        <param name="imu_udp_port" type="int" value="$(arg imu_udp_port)"/>                 
        <param name="imu_latency_microsec" type="int" value="0"/>                           
        <param name="imu_topic" type="string" value="$(arg imu_topic)"/>                    
        

please help me

rostest, 2024-12-12:
Thanks for your feedback. Did you get any warnings after starting sick_scan_xd? If so, please post a logfile or screenshot of any messages. Are the point clouds published and visible with rviz2?

To analyse missing IMU data, please check the IMU and network settings with SOPAS Air, i.e. open the lidar IP address 107.125.137.108 in a browser, switch to authorised client and check the IMU configuration. Does the IMU work correctly with SOPAS Air?

vanbacvb, 2024-12-12:
I got cloud data.

vanbacvb, 2024-12-12:
where is imu configuration?
[image]

vanbacvb, 2024-12-12:
I got cloud data. where is imu configuration?

rostest, 2024-12-16:
Thanks for your reply. The IMU configuration should be displayed in SOPAS Air in configuration/basic-settings and application/dataoutput:

[image]
[image]
[image]

If SOPAS Air does not show any IMU settings, please update the multiScan firmware to the latest release 2.3.0 and try again.

For further analysis please post a logfile or screenshots of the sick_scan_xd console output, especially the messages with SOPAS IMU requests and responses like "sWN ImuDataEnable ...".

closed 2024-12-20 by rostest