SICKAG/sick_scan_xd #436: Why do these problems occur?(Multiscan136)

goodmgk, 2024-12-19:
[image]
[image]
[image]

I tried my best to follow the manual but couldn't solve it.

rostest, 2024-12-19:
Thanks for your feedback. Parameter `udp_receiver_ip` must be set to the IP address of your PC to receive scan data from a multiScan lidar.

multiScan and picoScan lidars communicate via TCP, but send their scan data via UDP to the receiving PC (i.e. the sick_scan_xd driver). Therefore two IP addresses must be configured:
* `hostname:=<ip address of the multiScan lidar>`
* `udp_receiver_ip:=<ip address of the PC running sick_scan_xd>`

Specify both ip addresses in the launch file `sick_multiscan.launch` or by command line arguments `hostname:=<ip address of the multiScan lidar> udp_receiver_ip:=<ip address of the PC running sick_scan_xd>`, e.g. `hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.100`

Make sure that the multiScan and the PC have ip addresses in the same subnet. See [picoScan and multiScan configuration](https://github.com/SICKAG/sick_scan_xd#configuration) for examples.

To find out the IP address of your PC, run `ifconfig -a` on Linux or `ipconfig /all` on Windows. Alternatively, run SOPAS Air, i.e. open the lidars ip address in a browser (192.168.0.1 by default), switch to "authorized client" and select the network configuration and ip settings.

goodmgk, 2024-12-20:
First of all, thank you for your reply, I think I tried everything you suggested.
I will share the data I set up, so if there is anything wrong, please reply.

_**I am currently testing on Linux Ubuntu using wsl2 on Windows. Please note.**_

**(1) Multiscan setting**
[image]
[image]
[image]

**(2) PC setting** 
[image]

**(3) Linux ifconfig**
[image]

**(4) launch file : sick_multiscan.launch**
[image]

**(5) execution**
[image]

**(6) error occurred**
[image]

rostest, 2024-12-20:
Thanks for your reply and screenshots. I assume that `192.168.0.100` is the IP address of your Windows PC, but the Linux subsystem (WSL) running sick_scan_xd has the IP address `172.22.216.6`. To receive scan data via UDP, the Linux subsystem should have a static IP address in the same subnet, i.e. `192.168.0.x`. Please assign a static IP address `192.168.0.x` to your Linux subsystem. See for example [Assigning Static IP Addresses in WSL2](https://gist.github.com/wllmsash/1636b86eed45e4024fb9b7ecd25378ce) and [Make IP address of WSL2 static](https://superuser.com/questions/1582234/make-ip-address-of-wsl2-static/1715206#1715206) for more information.

Then start a browser such as firefox in the Linux subsystem (WSL), open `192.168.0.1` (the multiScan IP address) in that browser and use SOPAS Air to configure the network settings. If you can see the point cloud in SOPAS Air within your Linux subsystem, use these settings resp. IP addresses with sick_scan_xd.

rostest, 2025-01-07:
Closed due to inactivity.