SICKAG/sick_scan_xd #438: Some questions about TiM240

Jonny0816, 2025-01-26:
Dear Sir

Customer use the latest version driver with TiM240. it may need longer time to start up or report error when binary protocal set to true. After reboot can use normally. If set up the binary protocal to false, it can run well but the output angle seems not right and need to do a rotation.
Is there any comments why have these question? 
And why the binary protocal is recommended? what is the benifit?

Thanks.

rostest, 2025-01-27:
Thank you for your questions, which we will answer as follows:

* Start with binary protocol: Starting with the binary protocol will probably take longer as the device is configured to ASCII protocol by default. With SOPAS ET you can switch the device permanently to binary and write this setting to the EEPROM.
* Preference for binary protocol: The binary protocol allows compact storage of numerical values (such as distances). It is therefore recommended for better transmission of high-volume messages. In addition, not all lidars support the ASCII protocol.
* Differences between ASCII protocol and binary protocol: If you see discrepancies here, I would ask you to attach screenshots & the recording of startup sequences of the driver here.

rostest, 2025-02-16:
Closed due to inactivity.

closed 2025-02-16 by rostest