SICKAG/sick_scan_xd #440: tick_to_timestamp_mode multiscan 165

mrvsht, 2025-01-31:
## Problem
We use the multiscan 165 with this driver and have the following issue:
The tick_to_timestamp_mode is not working as expected. 
If we use the default value tick_to_timestamp_mode = 0, we get timestamps that looks correct.
With the value tick_to_timestamp_mode =  1 , we get timestamps that are on a fix value like: sec: -140462612 , nsec: 0. And with the value tick_to_timestamp_mode = 2 we get also timestamps that are on a fix value like: sec: 0 , nsec: 0.

As we understand it, the tick_to_timestamp_mode 1 and 2 should contain the hardware time of the scanner in the header stamp and not looks like described above.

## Our Setup
- Sick Scanner: multiScan165
- Driver: sick_scan_xd
- Firmware: 2.2.2
- ROS2: Humble
- Timesync: PTP

rostest, 2025-02-02:
Please use de Branch feature/issue_update_2412xx to fix this issue.

This behavior should not occur in this branch. This branch is also the basis for the next release.

closed 2025-02-02 by rostest

mrvsht, 2025-02-04:
Thanks this works