SICKAG/sick_scan_xd #441: timestamps with offset in field_t overlaps

mrvsht, 2025-01-31:
## Problem
We use the multiscan 165 with this driver and have the following issue:
We use the tick_to_timestamp_mode = 0 and get timestamps that look correct, but they overlap if you look closely at the t field.

If you calculate the offset in field t to each point from one scan, it looks like, that the latest points are detected, after the next scan is already started.
It looks like an overlap from one scan to the next scan.

We analysed the timestamps with the calculated offset in field t and tried to displayed it on the following image:
In the image you can see the header stamp as the dashed vertical line and the horizontal lines show the offsets for each layer.
You can see that some scans do not start exactly after the previous one according to the header stamp and the next, subsequent scan starts before the previous one is finished according to the header stamp.
It looks to us as if the headerstamps jump back and forth. 

[image]

Hopefully the picture helps to understand the problem.

## Our Setup
- Sick Scanner: multiScan165
- Driver: sick_scan_xd
- Firmware: 2.2.2
- ROS2: Humble
- Timesync: PTP

rostest, 2025-02-03:
It is important to understand the measurement principle of the multiScan in order to be able to interpret the time stamps for the multiScan. As you can see from the drawing below, the full scan is made up of 12 segments of 30 degrees each. Each layer in a segment has a different start azimuth angle as shown in the drawing. When 12 consecutive segments have been collected, a 360 degree scan can be output. This explains why the time stamps of the individual layers are sometimes offset by up to half a rotation time (i.e. 180 degrees).

Image taken from https://cdn.sick.com/media/docs/3/73/273/operating_instructions_multiscan165_en_im0110273.pdf

[image].

This behaviour is explained by the metrological setup shown in the drawing below (also taken from the same document).

You can see that 8 laser diodes are arranged around the red circle. Another measuring module is also mounted 180 degrees azimuthally and has the same structure. This generates another 8 layers at the same time. In total, all 16 layers are acquired simultaneously. However, each vertical layer covers a 30 degree azimuth range with a different azimuth start angle per layer.

It may be useful for your application to evaluate the 30 degree segments directly instead of waiting for the full scan (combined from 12 segments) (see "scan_segment" in the launch files).

[image].

mrvsht, 2025-02-04:
To use 30 degrees segments directly is definitely something we will be looking at.
We have just discovered that this error does not occur if we use tick_to_timestamp_mode 2, i.e. use the timestamp of the scanner.
From this I would conclude that this behavior is rather caused by the software PLL.
But this is only a vague assumption.
For us it helped to set tick_to_timestamp_mode 2.

rostest, 2025-02-09:
Thank you for your positive reply. Now that you have a working solution, we will close the ticket.

closed 2025-02-09 by rostest