SICKAG/sick_scan_xd #443: Low remission filter in c++ API for Picoscan150

fra222, 2025-02-04:
I am working with the 2D LIDAR sensor Picoscan150 in C++. How do I set the filter for "low remission"?

Thanks

rostest, 2025-02-09:
Thanks for your feedback. Not all available filter options can be configured by launchfile. We recommend to set the filter for low remission with SOPAS Air and to store the new settings in the lidars EEPROM. Switch to "Authorized Client" to modify the filter configuration.

Alternatively, you can send the corresponding SOPAS command to the lidar using API function SickScanApiSendSOPAS following this example:
```
// Send SOPAS command "sWN SensitivityMode 1" (default sensitivity) and receive SOPAS response
char sopas_response_buffer[1024] = { 0 };
if (SickScanApiSendSOPAS(apiHandle, "sWN SensitivityMode 1", &sopas_response_buffer[0], (int32_t)sizeof(sopas_response_buffer)) != SICK_SCAN_API_SUCCESS)
{
    printf("## SickScanApiSendSOPAS failed\n");
}
```

rostest, 2025-03-19:
Closed due to inactivity.

closed 2025-03-19 by rostest