SICKAG/sick_scan_xd #444: SOPAS start up sequence sFA errors

shubho-upenn, 2025-02-04:
Hi
I am trying to set up the drivers for SICK multiScan136 LiDAR with ROS. I am receiving SOPAS sFA errors. These errors occur during the SOPAS start up sequence, specifically during the sWN FREchoFilter 2, sWN ScanDataEthSettings 1, and sWN ScanDataEnable 0 messages. I am attaching the terminal output for the same below.

```
$ roslaunch sick_scan_xd sick_multiscan.launch hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.135

... logging to /home/shubhodeep/.ros/log/95003e36-e344-11ef-a33d-d5a4e60faf4d/roslaunch-shubhodeep-G15-44368.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://shubhodeep-G15:34155/

SUMMARY
========

PARAMETERS
 * /multiScan/add_transform_check_dynamic_updates: False
 * /multiScan/add_transform_xyz_rpy: 0,0,0,0,0,0
 * /multiScan/check_udp_receiver_ip: True
 * /multiScan/check_udp_receiver_port: 2116
 * /multiScan/client_authorization_pw: F4724744
 * /multiScan/cloud_all_fields_fullframe: coordinateNotatio...
 * /multiScan/cloud_all_fields_segments: coordinateNotatio...
 * /multiScan/cloud_polar_unstructured_fullframe: coordinateNotatio...
 * /multiScan/cloud_polar_unstructured_segments: coordinateNotatio...
 * /multiScan/cloud_structured: coordinateNotatio...
 * /multiScan/cloud_structured_hires0: coordinateNotatio...
 * /multiScan/cloud_structured_hires0_segments: coordinateNotatio...
 * /multiScan/cloud_structured_hires1: coordinateNotatio...
 * /multiScan/cloud_structured_hires1_segments: coordinateNotatio...
 * /multiScan/cloud_structured_segments: coordinateNotatio...
 * /multiScan/cloud_unstructured_echo1: coordinateNotatio...
 * /multiScan/cloud_unstructured_echo1_segments: coordinateNotatio...
 * /multiScan/cloud_unstructured_echo2: coordinateNotatio...
 * /multiScan/cloud_unstructured_echo2_segments: coordinateNotatio...
 * /multiScan/cloud_unstructured_echo3: coordinateNotatio...
 * /multiScan/cloud_unstructured_echo3_segments: coordinateNotatio...
 * /multiScan/cloud_unstructured_fullframe: coordinateNotatio...
 * /multiScan/cloud_unstructured_reflector: coordinateNotatio...
 * /multiScan/cloud_unstructured_reflector_segments: coordinateNotatio...
 * /multiScan/cloud_unstructured_segments: coordinateNotatio...
 * /multiScan/custom_pointclouds: cloud_unstructure...
 * /multiScan/export_csv: False
 * /multiScan/export_udp_msg: False
 * /multiScan/host_FREchoFilter: 2
 * /multiScan/host_LFPangleRangeFilter: 0 -180.0 +179.0 -...
 * /multiScan/host_LFPintervalFilter: 0 1
 * /multiScan/host_LFPlayerFilter: 0 1 1 1 1 1 1 1 1...
 * /multiScan/host_read_filtersettings: True
 * /multiScan/host_set_FREchoFilter: True
 * /multiScan/host_set_LFPangleRangeFilter: False
 * /multiScan/host_set_LFPintervalFilter: False
 * /multiScan/host_set_LFPlayerFilter: False
 * /multiScan/hostname: 192.168.0.1
 * /multiScan/imu_enable: True
 * /multiScan/imu_latency_microsec: 0
 * /multiScan/imu_topic: imu
 * /multiScan/imu_udp_port: 7503
 * /multiScan/laserscan_layer_filter: 0 0 0 0 0 1 0 0 0...
 * /multiScan/logfolder: 
 * /multiScan/measure_timing: True
 * /multiScan/msgpack_output_fifolength: 20
 * /multiScan/msgpack_validator_azimuth_end: 180.0
 * /multiScan/msgpack_validator_azimuth_start: -180.0
 * /multiScan/msgpack_validator_check_missing_scandata_interval: 12
 * /multiScan/msgpack_validator_discard_msgpacks_out_of_bounds: True
 * /multiScan/msgpack_validator_elevation_end: 90.0
 * /multiScan/msgpack_validator_elevation_start: -90.0
 * /multiScan/msgpack_validator_enabled: False
[... 263 lines cut ...]
[ INFO] [1738706918.190959665]: sick_generic_laser: stop and exit (line 218)
[ INFO] [1738706918.195351189]: sick_scansegment_xd finishing (382)
[ INFO] [1738706918.196593083]: sick_scansegment_xd finishing (392)
sick_scansegment_xd exit: sending stop commands...
[ INFO] [1738706918.196651809]: SickScanServices: Sending request "sMN SetAccessMode 3 F4724744"
[ INFO] [1738706918.196686596]: Sending  : <STX>sMN SetAccessMode 3 F4724744<ETX>
[ INFO] [1738706918.197338317]: Receiving: <STX>sAN SetAccessMode 0<ETX>
[ERROR] [1738706918.197437905]: Unexpected response "\x02sAN SetAccessMode 0\x03" received from lidar, "sAN SetAccessMode 0" not expected, SOPAS command failed.
[ INFO] [1738706918.197464840]: SickScanServices: Request "sMN SetAccessMode 3 F4724744" successfully sent, received reply "sAN SetAccessMode 0"
[ INFO] [1738706918.197486871]: SickScanServices: request: "sMN SetAccessMode 3 F4724744"
[ INFO] [1738706918.197503948]: SickScanServices: response: "sAN SetAccessMode 0"
[ INFO] [1738706918.197530940]: SickScanServices: Sending request "sWN ScanDataEnable 0"
[ INFO] [1738706918.197552451]: Sending  : <STX>sWN ScanDataEnable 0<ETX>
[ INFO] [1738706918.198250518]: Receiving: <STX>sFA A<ETX>
[ WARN] [1738706918.198296624]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX>sWN ScanDataEnable 0<ETX>, received answer: "sFA A", expected patterns: "sWA ScanDataEnable","sAN ScanDataEnable"
[ WARN] [1738706918.198311828]: Sopas error code A received from lidar
[ERROR] [1738706918.198326323]: ## ERROR SickScanServices::sendSopasAndCheckAnswer: error sending sopas command "sWN ScanDataEnable 0"
[ERROR] [1738706918.198338038]: ## ERROR SickScanServices::sendSopasCmdCheckResponse() failed on sending command"sWN ScanDataEnable 0"
[ERROR] [1738706918.198352905]: ## ERROR SickScanServices::sendMultiScanStopCmd(): sendSopasCmdCheckResponse("sWN ScanDataEnable 0") failed.
```

I have already cross-checked the <udp_receiver_ip> IP address and am able to access the LiDAR point cloud in the SOPAS web browser GUI. I attempted the setup in ROS1 (Noetic) with Ubuntu20.04 and ROS2 (Humble) with Ubuntu22.04, and am getting the same issue with both. 
The LiDAR is currently on firmware version 1.2.3.

Thanks in advance!

rostest, 2025-02-09:
Thanks for your feedback and logfile. The error message `Unexpected response "\x02sAN SetAccessMode 0\x03" received from lidar` indicates that the authorization with the default password failed. The missing authorization causes sFA errors on the next commands. Most probably the password was changed.

There are two options to solve the problem:

1. Reset the lidar to factory defaults. After that, the password is reset to the default value.

2. Determine the hash value of your password with SOPAS ET. The password hash is a 32 bit value, F4724744 by default. Set the new hash value in the launchfile sick_multiscan.launch:
    ```
    <param name="client_authorization_pw" type="string" value="F4724744" /> 
    ```

shubho-upenn, 2025-02-12:
Worked! Thanks for the help.

closed 2025-02-12 by shubho-upenn