SICKAG/sick_scan_xd #445: Publishing Frequency Drops to 1 Hz When Using SICK Picoscan150 Prime with IMU

Kyros8712, 2025-02-10:
We are using the SICK Picoscan150 Prime version along with an IMU. 
We have observed that the publishing frequency occasionally drops to 1 Hz.
Is there any way to improve or stabilize the publishing frequency?

[image]

gschwsiSICKAG, 2025-02-11:
Hi @Kyros8712, does this issue mainly occur after startup? If possible, you can record a Wireshark log of the UDP stream between the picoScan150 and your host system. This will help us determine whether the picoScan150 is not sending the packets or if the root cause is on the host side.

Kyros8712, 2025-02-14:
When the issue was detected, the cartographer node was also running.
We compared the time when the cartographer node displayed error messages to capture information from Wireshark.

wireshark
[image]
topic frequency
[image]
cartographer error message
[image]

Kyros8712, 2025-02-17:
The following link contains the data I obtained using Wireshark:

https://drive.google.com/drive/folders/1rWk5lRBqGhkfQO9BV9QrH42busG8g2kx?usp=drive_link

I first enabled Wireshark, then used sick_picoscan.launch to start picoscan150, followed by launching Cartographer.

The time when sick_picoscan.launch was started:

[image]

The time when Cartographer issued a warning about IMU data stopping updates:

[image]

[image]

Using the sick_picoscan.launch start time as 0, the IMU data anomaly was detected approximately 523 seconds later:

1739780732.37860192523.59160 − 1739780208.7870243523.591626 = 523.591577594

weinmalSICKAG, 2025-02-20:
Dear @Kyros8712,
I am continuing the inverstigation into this issue. Due to our company network policies, I am not allowed to visit Google Drive. Could you please upload the data optained via Wireshark to Secureshare? https://secureshare.sick.com/Start?0
During the upload process you are asked to which email Addresses you want to send the data. Please enter: alexander.weinmann@sick.de

Kyros8712, 2025-02-21:
OK! Thanks for your help!

weinmalSICKAG, 2025-02-21:
Thank you for providing the file. After anaylsing the network traffic this seems to be an issue with the device it self and not sick_scan_xd. I will contact you via email to resolve the issue with the device.

closed 2025-02-21 by weinmalSICKAG