SICKAG/sick_scan_xd #446: Hello! I have a question! Hadware version.

wanilly, 2025-02-13:
we use CM5(Raspberry Pi) device. By the way, can we use sick Lidar api(sick scan xd: TiM7xxS) for this device? 

Our system specifications are as follows:
- OS: Ubuntu24.10 (Raspberry Pi)
- Architecture: AARCH64 (ARM64)
- Kernel: Linux 6.11.0-1007-raspi

Thank for your response...

wanilly, 2025-02-14:
Thank you for your response!!😃😃

TienYuWu, 2025-02-18:
I have a similar issue. Does S30B-2011DA be supported by this package?

wanilly, 2025-02-25:
hello, I have a problem for build on CM5(OS: Ubuntu24.10 (Raspberry Pi)
why occure this problem? In other device, It works fine with other devices.... plz... 
[image]

## WARNING SickScanApiLoadLibrary: LoadLibrary("libsick_scan_xd_shared_lib.so") failed
## ERROR SickScanApiCreate: library not loaded
## ERROR SickScanApi, cacheFunctionPtr(SickScanApiInitByCli): library not initialized
## ERROR SickScanApi, cacheFunctionPtr(SickScanApiInitByCli): GetProcAddress failed
## ERROR SickScanApiInitByCli: library call SickScanApiInitByCli() failed, error code 3
## ERROR SickScanApi, cacheFunctionPtr(SickScanApiRegisterCartesianPointCloudMsg): library not initialized
## ERROR SickScanApi, cacheFunctionPtr(SickScanApiRegisterCartesianPointCloudMsg): GetProcAddress failed
## ERROR SickScanApiRegisterCartesianPointCloudMsg: library call SickScanApiRegisterCartesianPointCloudMsg() failed, error code 3
^Z

rostest, 2025-03-19:
Closed due to inactivity.

closed 2025-03-19 by rostest