SICKAG/sick_scan_xd #448: Any dedicated interpreter available for sick lidars to run .pcap files?

purpleLightningG, 2025-02-17:
Hi, I am using a sick lidar and recorded data in .pcd, .bag and .pcap format. I downloaded LidarView in my mac to view the .pcap file but there seems to be no interpreter available for sick lidars

[image]

My question is, are there any interpreters for visualizing the .pcap files for sick lidars or any dedicated application for sick lidars to visualize the .pcap files? Thanks in advance.

rostest, 2025-02-18:
We guess that you are referring to this repo:

https://github.com/Kitware/LidarView

This repository does not currently support the SICK lidars. “pcap” files are recorded with WireShark. When an interpreter for “LidarView” reads this data, it decodes the IP traffic. The actual payload of the data is stored in SICK's own SOPAS data protocol embedded in IP data stream.

In the graphic [image] you can see that you need an interpreter to decode the raw device packets.  This interpreter for SOPAS is currently not available for a SICK device.

purpleLightningG, 2025-02-18:
Thanks for your detailed response. I am currently using open3D to visualize the pcd files. Are there any other visualizers available or something that you recommend other than open3D to play bag/pcd data files produced by sick_scan lidars? Thanks in advance.

weinmalSICKAG, 2025-02-20:
We recommend to view the .pcd files with CloudCompare: https://www.danielgm.net/cc/

purpleLightningG, 2025-02-23:
Thanks a lot!

closed 2025-02-23 by purpleLightningG