SICKAG/sick_scan_xd #449: sick_generic_laser: leaving main loop...[Info]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).

Abdulhadiasa, 2025-02-18:
I am running the API (no ROS) and I added a deep copy (using copy.deepcopy) of the cloud points before any external processing, yet I keep getting the following error after running for a while
```
sick_generic_laser: leaving main loop...[Info]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).
SickScanCommon closed.
[Info]: Caught signal 2

[Info]: good bye

[Info]: You are leaving the following version of this node:

[Info]: sick_scan_api V. 3.6.0 githash:"7cab14fb4e3ec05fb59eea19039823510abfc688" gitinfo:"2024-10-23 14:46:37 +0200	Alexander Weinmann	Merge branch 'release/3.6.0'"

[Info]: sick_generic_laser: stop and exit (line 218)
[Info]: sick_generic_laser: exit (line 221)
sick_generic_laser: exit (line 223)
sick_generic_laser: exit (line 225)

```
How can I debug this?

here are my run parameters:
```
[Info]: sick_scan_api V. 3.6.0 githash:"7cab14fb4e3ec05fb59eea19039823510abfc688" gitinfo:"2024-10-23 14:46:37 +0200	Alexander Weinmann	Merge branch 'release/3.6.0'"
[Info]: SickScanApiInitByLaunchfile: launchfile_args = "sick_mrs_1xxx.launch hostname:=192.168.5.53"
[Info]: SickScanApiInitByCli:  sick_mrs_1xxx.launch hostname:=192.168.5.53
[Info]: Try loading launchfile : sick_mrs_1xxx.launch
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["hostname"]="192.168.0.1"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["cloud_topic"]="cloud"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["laserscan_topic"]="scan"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["frame_id"]="cloud"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["sw_pll_only_publish"]="true"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["imu_enable"]="true"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["imu_topic"]="imu"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["imu_frame_id"]="imu_link"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["nodename"]="sick_mrs_1xxx"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["add_transform_xyz_rpy"]="0,0,0,0,0,0"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["add_transform_check_dynamic_updates"]="false"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["ang_res"]="0.0625"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["scan_freq"]="50"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["scan_layer_filter"]="4 1 1 1 1"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): default_args["tf_publish_rate"]="10.0"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="scanner_type", type="string", value="sick_mrs_1xxx"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="min_ang", type="double", value="-1.570796327"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="max_ang", type="double", value="+0.872664626"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="intensity_resolution_16bit", type="bool", value="false"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="hostname", type="string", value="$(arg hostname)" overwritten by default value "192.168.0.1"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="hostname", type="string", value="192.168.0.1"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="port", type="string", value="2112"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="timelimit", type="int", value="5"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="cloud_topic", type="string", value="$(arg cloud_topic)" overwritten by default value "cloud"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="cloud_topic", type="string", value="cloud"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="laserscan_topic", type="string", value="$(arg laserscan_topic)" overwritten by default value "scan"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="laserscan_topic", type="string", value="scan"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="frame_id", type="str", value="$(arg frame_id)" overwritten by default value "cloud"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="frame_id", type="str", value="cloud"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="sw_pll_only_publish", type="bool", value="$(arg sw_pll_only_publish)" overwritten by default value "true"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="sw_pll_only_publish", type="bool", value="true"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="use_generation_timestamp", type="bool", value="false"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="min_intensity", type="double", value="0.0"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="scandatacfg_timingflag", type="int", value="-1"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="scandatacfg_azimuth_table", type="int", value="0"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="imu_enable", type="bool", value="$(arg imu_enable)" overwritten by default value "true"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="imu_enable", type="bool", value="true"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="imu_enable_additional_info", type="bool", value="True"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="imu_topic", type="string", value="$(arg imu_topic)" overwritten by default value "imu"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="imu_topic", type="string", value="imu"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="range_min", type="double", value="0.05"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="range_max", type="double", value="100.0"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="range_filter_handling", type="int", value="0"
[Info]: LaunchParser::parseFile(sick_mrs_1xxx.launch): name="ang_res", type="double", value="$(arg ang_res)" overwritten by default value "0.0625"
[... 195 lines cut ...]
[Info]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[Info]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[Info]: Waiting for scanner ready state since 11 secs
[Info]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[Info]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[Info]: Waiting for scanner ready state since 12 secs
[Info]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[Info]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[Info]: Waiting for scanner ready state since 13 secs
[Info]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[Info]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[Info]: Waiting for scanner ready state since 14 secs
[Info]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[Info]: Receiving: <STX>sRA SCdevicestate \x01<ETX>
[Info]: Scanner ready for measurement after 15 [sec]
[Info]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[Info]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[Info]: Sending  : <STX><STX><STX><STX><Len=0029>sEN InertialMeasurementUnit 0x01 CRC:<0x35>
[Info]: Receiving: <STX>sEA InertialMeasurementUnit \x01<ETX>
[Info]: 1 / 6 packets dropped. Software PLL not yet locked.
[Info]: 2 / 6 packets dropped. Software PLL not yet locked.
[Info]: 3 / 6 packets dropped. Software PLL not yet locked.
[Info]: 4 / 6 packets dropped. Software PLL not yet locked.
[Info]: 5 / 6 packets dropped. Software PLL not yet locked.
```

rostest, 2025-02-18:
Hello @Abdulhadiasa,

Thank you for reporting this issue and providing detailed information.

The message [Info]: Caught signal 2 indicates that the application received an interrupt signal (SIGINT), which typically corresponds to a manual interruption (e.g., pressing Ctrl+C) or an external process sending this signal.

Given that you're running the API without ROS and have implemented a deep copy of the point cloud data before external processing, it's possible that the interruption is due to one of the following reasons:

- Manual Interruption: Ensure that there isn't an inadvertent Ctrl+C input or another process intentionally sending a SIGINT to your application.

- Resource Constraints: Intensive processing of point cloud data can lead to high CPU or memory usage. If the system becomes resource-starved, it might terminate processes. Monitor your system's resource usage to identify any potential bottlenecks.

- Unhandled Exceptions: If an exception occurs during the processing of the point cloud data and isn't properly handled, it could cause the application to exit unexpectedly. Ensure that all potential exceptions are caught and managed appropriately.

To effectively debug the issue you're encountering with the `sick_scan_xd` API, consider the following steps:

## 1. Compile with Debug Information
- **Modify Compiler Flags:**  
  By default, `sick_scan_xd` builds with the `-O3` optimization flag. For debugging, it's beneficial to include debug symbols and reduce optimization levels. Adjust your build configuration to use `-g` for debug information and `-O0` to disable optimizations.
- **Rebuild the Project:**  
  After modifying the compiler flags, perform a clean rebuild of the project to ensure all components include the necessary debug information.

## 2. Utilize Debugging Tools
- **GDB (GNU Debugger):**  
  Run your application within GDB to capture the state at the time of the signal.
  - **Start GDB:**  
    ```bash
    gdb --args ./your_application [arguments]
    ```
  - **Run the Application:**  
    Within GDB, type `run` to start the program.
  - **Capture Backtrace on Signal:**  
    If the program exits due to a signal (e.g., SIGINT), GDB will pause execution. Type `bt` to obtain a backtrace, which can help identify where the signal was raised.

- **Signal Handling:**  
  Ensure that your application or any linked libraries aren't inadvertently sending SIGINT signals. Review the code for any `raise(SIGINT)` or similar calls.

## 3. Review Signal Handling in `sick_scan_xd`
- **Signal Setup:**  
  The `sick_scan_xd` driver sets up signal handlers for SIGINT and SIGTERM to manage graceful shutdowns. This setup is located in the `sick_generic_caller.cpp` file:
  ```cpp
  signal(SIGINT, rosSignalHandler); // SIGINT = 2, Ctrl-C or kill -2
  signal(SIGTERM, rosSignalHandler); // SIGTERM = 15, default kill level
Custom Signal Handling:
If your application modifies signal handling, ensure it doesn't conflict with the driver's handlers. Conflicting handlers might lead to unexpected terminations.

rostest, 2025-02-18:
For the most recent updates, please refer to the latest branch:
🔗 [SICK Scan XD - feature/issue_update_2412xx](https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx)

BTW:  the startup sequence looks good.

Abdulhadiasa, 2025-02-19:
I did not mention this before, I am using python in my application.
regarding your points:

> Manual Interruption: Ensure that there isn't an inadvertent Ctrl+C input or another process intentionally sending a SIGINT to your application.

There was no manual interruption

>Resource Constraints: Intensive processing of point cloud data can lead to high CPU or memory usage. If the system becomes resource-starved, it might terminate processes. Monitor your system's resource usage to identify any potential bottlenecks.

My CPU was under 50% utilization and there was more than enough RAM on my PC (unless you mean Lidar resources, let me know how I can check this)

>Unhandled Exceptions: If an exception occurs during the processing of the point cloud data and isn't properly handled, it could cause the application to exit unexpectedly. Ensure that all potential exceptions are caught and managed appropriately.

my application was still running after the Lidar socket was disconnected and just to be sure I surrounded my code with a general try/except but no exceptions were caught

**Compile with Debug Information**
------
I compiled with `cmake -DROS_VERSION=0 -DLDMRS=0 -DSCANSEGMENT_XD=0 -g -O0 -G "Unix Makefiles" ../sick_scan_xd` but no new debug messages were added when the Lidar stopped providing data (output will be added in the gdb section bellow)

Here is the output from gdb:
```
...
[Info]: 1 / 6 packets dropped. Software PLL not yet locked.
[Info]: 2 / 6 packets dropped. Software PLL not yet locked.
[Info]: 3 / 6 packets dropped. Software PLL not yet locked.
[Info]: 4 / 6 packets dropped. Software PLL not yet locked.
[Info]: 5 / 6 packets dropped. Software PLL not yet locked.
```
up to this point, I was receiving Lidar readings
```
sick_generic_laser: leaving main loop...[Info]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).
SickScanCommon closed.
[Thread 0x7fffc124a700 (LWP 1599052) exited]
[Thread 0x7fffc1b3d700 (LWP 1599051) exited]
```
after the above, my application was still running but the Lidar readings stopped, when i closed my application and called
```
SickScanApiUnloadLibrary(self.api_lib)
SickScanApiDeregisterCartesianPointCloudMsg(self.api_lib, self.api_handler, self.cartesian_pointcloud_callback)
SickScanApiClose(self.api_lib, self.api_handler)
SickScanApiRelease(self.api_lib, self.api_handler)
```
then the `signal 2` was caught
```
[Info]: Caught signal 2

[Info]: good bye

[Info]: You are leaving the following version of this node:

[Info]: sick_scan_api V. 3.6.0 githash:"7cab14fb4e3ec05fb59eea19039823510abfc688" gitinfo:"2024-10-23 14:46:37 +0200	Alexander Weinmann	Merge branch 'release/3.6.0'"

[Info]: sick_generic_laser: stop and exit (line 218)
[Info]: sick_generic_laser: exit (line 221)
sick_generic_laser: exit (line 223)
sick_generic_laser: exit (line 225)
[Thread 0x7fffd386c700 (LWP 1599050) exited]
[Thread 0x7fffd606d700 (LWP 1599049) exited]
[Thread 0x7fffd686e700 (LWP 1599048) exited]
...
```
and `bt` gave `No stack.`

Abdulhadiasa, 2025-02-19:
I tried running the API while simply opening camera stream (once using gstreamer, and another using opencv) without doing any processing on the cloud points except for the deep copy operation and got the same result
```
sick_generic_laser: leaving main loop...[Info]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).
SickScanCommon closed.
```

rostest, 2025-03-19:
It is not entirely clear to us whether you still have a problem here. We think that we should clear up a misunderstanding here. Debug’ does not refer to output from the driver. Rather, it means direct debugging of the C++ application. If you have no experience here as a Python developer, we recommend that you familiarise yourself with gdb here.

rostest, 2025-04-09:
Closed due to inactivity.

closed 2025-04-09 by rostest

spark-res, 2025-07-11:
Hi @Abdulhadiasa, did you ever find a resolution for this issue?

We had this issue earlier. [See thread](https://github.com/SICKAG/sick_scan_xd/issues/361#issuecomment-2290769241).

It was never fully resolved, it seems there's a few related issues about the place. Resolution seems to be hardware related or closed due to inactivity.

We are using Python, no ROS NVIDEA Jetson.