SICKAG/sick_scan_xd #451: C++ API for Picoscan150 init isuee

sisusch, 2025-02-20:
We received the lidar picoscan 150, but I'm having problems using the library or setting the .launch file (https://github.com/SICKAG/sick_scan_xd).

I followed the instructions reported into the to generate the dll etc, this was generated without problems, both the degug and the release, at the moment I'm using the debug.

Alternatively, follow the build and run instructions on Windows:

REM Set environment for Visual Studio 2019 (VS 16)
set _os=x64
set _cmake_string=Visual Studio 16
set _msvc=Visual Studio 2019
set _cmake_build_dir=build
REM Build the minimalistic C++ usage example
cd examples\cpp
mkdir %_cmake_build_dir%
cd %_cmake_build_dir%
cmake -G "%_cmake_string%" ..
cmake --build . --clean-first --config Debug
REM Set environment: add build folder to LD_LIBRARY_PATH, add python/api to PYTHONPATH
cd ..\..
set PATH=.;.\build;.\build\Debug;.\build_win64;.\build_win64\Debug;%PATH%
REM Run minimalistic C++ api example
.\examples\cpp\build\Debug\minimum_sick_scan_api_client.exe <launchfile> hostname:=<lidar-ip-address>

The sensor has a fixed address 192.168.1.200, while my pc is at 192.168.1.30.

I opened the port in my firewall at 2115, which is the same configured in the sensor and launchfile for data transmission.

In particular, the project used comes from the example c++ (minimum_sick_scan_api_client.cpp) project available within the library. Specifically, the problem/error occurs in the init phase of the sensor, with this error printed in CLI, as if it could not read the .launch configuration file properly, or there was an error in the c++ library.

[image]

This is my sick_picoscan.launch file configuration
[sick_picoscan.txt](https://github.com/user-attachments/files/18885310/sick_picoscan.txt)

Thanks for any support, if needed I can also share the complete vs2019 project.

sisusch, 2025-02-20:
I add a wireshark image that captures the frames coming from the sensor

[image]

rostest, 2025-02-20:
I recommend that you start with as simple an example as possible.

In this case, this means changing the original sick_picoscan.launch as little as possible.

In the original file, for example, the scanner type is “sick_picoscan”. This is not the case in your example. This is why this error is generated. You can also adjust the host name, as you are using 192.168.1.200 for the Lidar.
You can pass the hostname via CLI or in the launch file.

sisusch, 2025-02-21:
Thanks rostest,

I will start from the basic example without changes as suggested by you.
The change I made to the scanner type field is due to the suggestion given by CLI (last line).

[image]

regards.

sisusch, 2025-02-21:
Hello everyone for those who are in the same problem as above, I'll report the solution. 
As suggested, I started a new project, but printing the arguments passed from CLI, and this way I have a number greater than 1

[image]

Instead in my project (which was causing the error) I inserted the arguments directly into the code, such as the name of the launchfile and I launched the exde to debug directly from visual studio interface.

This leads to having only one argument and consequently in this portion of code, report below it is not executed. This unexecuted portion does not allow the configuration file loading to be load, causing the problem.

[image]

Thanks for the support.

closed 2025-02-21 by sisusch

rostest, 2025-02-24:
@sisusch 

We are very pleased that this issue has been resolved.
As an additional hint, we would like to provide the following information as general support:

Functions `SickScanApiCreate` and `SickScanApiInitByCli` expect arguments `argc` and `argv` as provided in the `main`. There are two alternatives to solve the problem:

1. Use the command line arguments: Pass `argc` and `argv` directly to `SickScanApiCreate` and `SickScanApiInitByCli`:
    ```
    SickScanApiHandle apiHandle = SickScanApiCreate(argc, argv);
    if (!apiHandle)
    {
        std::cerr << "Errore: impossibile creare l'handle della libreria!" << std::endl;
        return -1;
    }
    int result = SickScanApiInitByCli(apiHandle, argc, argv);
    if (result != 0) {
        std::cerr << "Errore nell'inizializzazione del LIDAR! Codice errore: " << result << std::endl;
        return -1;
    }
    ```
    Then call LIDAR_2D with launchfile and ip addresses in the command line:
    ```
    LIDAR_2D sick_picoscan.launch hostname:=192.168.1.200 udp_receiver_ip:=192.168.1.30
    ```

2. Provide lidar parameter directly using the same arguments:
    ```
    char* lidar_argv[] = { "LIDAR_2D", "sick_picoscan.launch", "hostname:=192.168.1.200", "udp_receiver_ip:=192.168.1.30" };
    int lidar_argc = 4;
    SickScanApiHandle apiHandle = SickScanApiCreate(lidar_argc, lidar_argv);
    if (!apiHandle)
    {
        std::cerr << "Errore: impossibile creare l'handle della libreria!" << std::endl;
        return -1;
    }
    int result = SickScanApiInitByCli(apiHandle, lidar_argc, lidar_argv);
    if (result != 0) {
        std::cerr << "Errore nell'inizializzazione del LIDAR! Codice errore: " << result << std::endl;
        return -1;
    }
    ```

Both alternatives are the same. Add a path to the launch file, if it is not in your current working directory, i.e. replace `sick_picoscan.launch` by `<path/to/>sick_picoscan.launch`. See [sick_scan_xd_api_test.cpp](https://github.com/SICKAG/sick_scan_xd/blob/develop/test/src/sick_scan_xd_api/sick_scan_xd_api_test.cpp) for an example.