SICKAG/sick_scan_xd #452: How to set maximum and minimum angles in sick_picoscan.launch driver

hoon0309, 2025-02-25:
I'm currently using picoscan 150 lidar. I've tried to build to version github 3.6.0. I'm using the file 'sick_picoscan.launch'. I tried changing several parameters, but 'fullframe_scan' on rviz was only scanning the same area. I'd like to know if there's a way to set the scan angle of lidar in the launch file or on another code.

[image]

rostest, 2025-02-25:
Please try to modify the following settings in your `sick_picoscan.launch` like the following example:
```
    host_LFPangleRangeFilter = "1 -30.0 +40.0 -90.0 +90.0 1"; // Optionally set LFPangleRangeFilter to "<enabled> <azimuth_start> <azimuth_stop> <elevation_start> <elevation_stop> <beam_increment>" with azimuth and elevation given in degree
    host_set_LFPangleRangeFilter = true;                       // If true, LFPangleRangeFilter is set at startup (default: false)
```
This should allow filtering to an azimuth angle in the range [-30 deg, 40 deg]. Please modify as needed.

hoon0309, 2025-02-26:
Thank you. I confirmed that it is the way you answered. However, when I set and executed as below in the launch file, there was no problem with angle_min value, but there was a problem with angle_max value of 160 degrees. Can you tell me what the problem is? I attached the picture as below.

[image]
[image]

weinmalSICKAG, 2025-02-26:
Please note that the picoScan has field of view of 276 °. Thus angle_min cannot be lower than -138° and angle_max cannot be larger than 138°. Does this explain your problem?
Also, the angle_min and angle_max that you can see in the lower screenshot are in radians. The angles that you configured in the .launch file are in degrees.

hoon0309, 2025-02-26:
Yes. Of course, I know that the value is radian, and I know that the maximum scan range is -138-138 degrees. That's why I asked you if there are cases where the scan range is out of the maximum value because when converting 2.821668 to degree, it comes out 161.66966759 degrees.

[image]

rostest, 2025-02-26:
First of all, we are pleased that you were able to make the adjustment. The reported angle difference between max_ang and min_ang is almost exactly 300 degrees. We will see if this is possibly due to the fact that a total of 10 segments of 30 degrees each are used to generate the output of the scan.

[image]

The drawing is taken from the document https://cdn.sick.com/media/docs/1/91/691/operating_instructions_picoscan150_2d_lidar_sensors_en_im0106691.pdf and shows that data from a total of 10 segments is used in this cycle. We will investigate whether the behavior you mentioned is possibly related to this. Since “min_ang” and “angle_increment” are suitable, you can use them to determine the angle values according to `min_ang + index * angle_increment` without any problems.

BTW: angle_increment 0.005817905534058809 rad is 0.3333 deg

weinmalSICKAG, 2025-02-26:
That was my oversight. You are right the angle_may is greater than what one would expect.
The reason for this is a bit involved. The picoScan sends the data for one full rotation in multiple segments. Each segments contains the same amunt of data. Depending on the preformance profile the segments cover either 30° or 60°. Lets assume that a performance profile is chosen that results in 60° segments. To cover the 276° FOV five 60° segments are required. The first segments starts at -138°. Thus the fifth segment starts at 102° (102°=-136° + 4*60°). The fifth segment ends at 162° which is what you see in the screenshots you postet. Since the fifth segment overlaps with the blindspot at the back of the picoScan, the data is padded with zeros to make the segment cover 60°.
The screenshot below shows this behaviour.
[image]
The driver collects these segments and provides a topic with the data of a full rotation. To filter out points with distance zero configure the rangeFilter in the .launch file.

[image]

FelixFrog, 2025-03-14:
Hi! I am facing the same issue using a picoScan 150. Since the last segment added is padded with zeros before being published to the scan_fullframe topic I was looking for a way to remove those zero points. I tried using 

```
<arg name="host_LFPangleRangeFilter" default="1 -135.0 +135.0 -90.0 +90.0 1" />
<arg name="host_set_LFPangleRangeFilter" default="True" />
```
which works but still zeroes the values instead of removing them. I tried searching for a way to enable `rangeFilter` with the `filter_flag` but it looks like it's used only in pointcloud definitions.

So is there a way to remove invalid points from the LaserScan messages and get a simmetrical pair of angle_min and angle_max?

weinmalSICKAG, 2025-03-24:
Thank you for your message.
We do not plan to fix the behaviour with the padded zeros. Downstream algorithms which consume the laser scan messages need to deal with invalid points anyway. For example consider beams which hit black targets at a great distance or beams which dont get any echo. These are also assigned a distance value of zero. In general beams which have a distance of less than lets say 30mm can be safely ignored because that would be a measurement inside the cover of the picoScan.

To answer the separate issue about the frameid: As a workaround you could add an identity transform from `myFrameId` to `myFrameId_1` as described here: https://github.com/SICKAG/sick_scan_xd/issues/434#issuecomment-2572782123

If you have any further questions regarding padded zeros don't hesitate to post them here. If you have other further questions please open another issue.

rostest, 2025-04-09:
Closed due to clarification.

closed 2025-04-09 by rostest