SICKAG/sick_scan_xd #453: problem running ros2 service call /ColaMsg

FL3UR-TOUW, 2025-02-28:
Hello guys me again,

Windows 11
Ros2 humble
sick_scan_xd
python 3.8.3

i am trying to run this command:  ros2 service call /ColaMsg sick_scan_xd/srv/ColaMsgSrv "{request: 'sEN LIDoutputstate 1'}"
while i am running this in an other cmd: ros2 launch sick_scan_xd sick_picoscan.launch.py hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.15 (this works fine, it shows the IMU data of the sensor).

but the ColaMsgSrv throws up error's were i can't get rid off!

c:\sick_scan_ws>ros2 service call /ColaMsg sick_scan_xd/srv/ColaMsgSrv "{request: 'sEN LIDoutputstate 1'}"
Traceback (most recent call last):
  File "C:\dev\ros2_humble\ros2-windows\Lib\site-packages\rosidl_generator_py\import_type_support_impl.py", line 46, in import_type_support
    return importlib.import_module(module_name, package=pkg_name)
  File "C:\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'sick_scan_xd.sick_scan_xd_s__rosidl_typesupport_c'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\dev\ros2_humble\ros2-windows\Scripts\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.11', 'console_scripts', 'ros2')())
  File "C:\dev\ros2_humble\ros2-windows\Lib\site-packages\ros2cli\cli.py", line 91, in main
    rc = extension.main(parser=parser, args=args)
  File "C:\dev\ros2_humble\ros2-windows\Lib\site-packages\ros2service\command\service.py", line 41, in main
    return extension.main(args=args)
  File "C:\dev\ros2_humble\ros2-windows\Lib\site-packages\ros2service\verb\call.py", line 58, in main
    return requester(
  File "C:\dev\ros2_humble\ros2-windows\Lib\site-packages\ros2service\verb\call.py", line 86, in requester
    cli = node.create_client(srv_module, service_name)
  File "C:\dev\ros2_humble\ros2-windows\Lib\site-packages\rclpy\node.py", line 1413, in create_client
    check_is_valid_srv_type(srv_type)
  File "C:\dev\ros2_humble\ros2-windows\Lib\site-packages\rclpy\type_support.py", line 51, in check_is_valid_srv_type
    check_for_type_support(srv_type)
  File "C:\dev\ros2_humble\ros2-windows\Lib\site-packages\rclpy\type_support.py", line 29, in check_for_type_support
    msg_or_srv_type.__class__.__import_type_support__()
  File "c:\sick_scan_ws\install\sick_scan_xd\Lib\site-packages\sick_scan_xd\srv\_cola_msg_srv.py", line 259, in __import_type_support__
    module = import_type_support('sick_scan_xd')
  File "C:\dev\ros2_humble\ros2-windows\Lib\site-packages\rosidl_generator_py\import_type_support_impl.py", line 48, in import_type_support
    raise UnsupportedTypeSupport(pkg_name)
rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'sick_scan_xd'

does someone know what i can do about this?

weinmalSICKAG, 2025-03-20:
Hello FL3UR-TOUW,
thank you for your message.
This issue might occur because you changed from the debug version of Python to the non-debug version of Python and some packages were only installed for the debug version. You might try to fix this by looking for a file called `sick_scan_xd_s__rosidl_typesupport_c_d.pyd` in `C:\dev\ros2_humble\ros2-windows\Lib\site-packages\sick_scan_xd`. Copy and rename that file to `sick_scan_xd_s__rosidl_typesupport_c.pyd`. Now the non-debug version of Python should be able to find it.
Let me know if that helped or if you need further assistance.

FL3UR-TOUW, 2025-03-20:
I found another solution, in the Cmake list of the sick_scan_xd package before building i changed this line: 

# Debug or Release build options
option(BUILD_DEBUG_TARGET "Build debug target (ON) or release (OFF)" OFF) # OFF (release) or ON (development)

So it builds in release instead of debug mode, then everything works as it should.

closed 2025-03-20 by weinmalSICKAG