SICKAG/sick_scan_xd #454: No full frame picoscan 150, from callback

sisusch, 2025-02-28:
Hello everyone, 
I would like to collect the data that I receive from the various frames generated by the picoscan 150 sensor.
At the moment is set to 15hz, 0.05° angle step in compact format udp.
In particular, I am only interested in collecting the fullframe with all the points and not the partial segmented ones. I followed the various posts, but I do not receive any element with segment id -1, when the call back is called.

Below I report the code in which I inserted a brake point inside the if, but this is never triggered.
In attachment to wireshark sniff, the CLI reply and the launchfile configuration.

Kind regards

```
void customizedPointCloudMsgCb(SickScanApiHandle apiHandle, const SickScanPointCloudMsg* msg)
{
	//std::cout << "C++ PointCloudMsgCb: " << msg->width << " x " << msg->height << " pointcloud message received" << std::endl;

	// Calcolare la dimensione totale del buffer da ricostruire
	size_t total_size = msg->row_step * msg->height;  // Dimensione totale del buffer (in byte)

	// Alloca il buffer completo per contenere tutti i dati e fare una hard copy prima che un altro settori sia pronto
	uint8_t* full_buffer = new uint8_t[total_size];

	// Variabile per tracciare la posizione corrente nel buffer
	size_t current_index = 0;
	if (msg->segment_idx == -1)
	{
		std::cout << "Segment number:" << msg->segment_idx << std::endl;
	}

}
```

[sick_picoscan no full frame.zip](https://github.com/user-attachments/files/19030054/sick_picoscan.no.full.frame.zip)

[no_full_frame.txt](https://github.com/user-attachments/files/19030024/no_full_frame.txt)

rostest, 2025-03-01:
Hello,

You have changed many entries compared to the reference launch file. We generally cannot recommend this. In particular, the entry `<param name=“check_udp_receiver_port” type=“int” value=“2116” /> ` must not be changed to 2115. 

Please find attached a working launch file derived from the template you sent. This will receive full scans.

You can then develop further from there.

[sick_picoscan.zip](https://github.com/user-attachments/files/19039284/sick_picoscan.zip)

sisusch, 2025-03-03:
Good morning Rostest, 
thanks for the suggestion I will make this change to my launch file.
In order to make all the changes suggested, I would ask you if you can attach the launch file you modified, because in the attached compressed folder there is only a file with a wireshark acquisition.

thanks

rostest, 2025-03-03:
[sick_picoscan.zip](https://github.com/user-attachments/files/19049663/sick_picoscan.zip)

Please find attached the zip file containing the launch file from our tests.

closed 2025-03-04 by sisusch