SICKAG/sick_scan_xd #456: PicoScan150 Prime data transmission occasionally stutters when using performanceprofilenumber=10

leonvictor, 2025-03-06:
Hi,

My application uses the C++ api to communicate with a PicoScan150 Prime over ethernet, using the callback approach. I'm receiving data as expected in pretty much all configurations.

The only exception is when using the `performanceprofilenumber=10` option (40Hz/0.125deg). In that case, every few minutes, the callback is not called for ~5s, then everything goes back to normal.

Here's all my customized configuration parameters:
```
hostname:=<device hostname>
udp_receiver_ip:=<receiver hostname> 
udp_port:=43871 
imu_udp_port:=46263 
performanceprofilenumber:=10 
host_FREchoFilter:=1 
custom_pointclouds:=cloud_unstructured_fullframe
```

I figure it might be a buffering issue somewhere, but I'm running out of ideas. Have you experienced such situations ? Do you have any pointers to potential solutions ?

Thanks for your help

gschwsiSICKAG, 2025-03-07:
Hi @leonvictor,

To rule out device-side issues, could you check if UDP messages are still being sent from the picoScan to your PC (e.g. with Wireshark) during these missing callback periods? If they are, we can then investigate potential buffering issues or other root causes.

rostest, 2025-04-09:
Hello @leonvictor, we are closing this ticket due to inactivity. If there are any updates on this issue, please feel free to reopen it.

closed 2025-04-09 by rostest