SICKAG/sick_scan_xd #460: `std::runtime_error` - Cannot store a negative time point in `rclcpp::Time

asokolowski-it, 2025-03-14:
## Description
When running `sick_scan_xd` with ROS 2 Humble on Ubuntu 22.04, the node terminates after some time with the following error:

```sh
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot store a negative time point in rclcpp::Time
[ros2run]: Aborted
```

## Environment
- **Operating System**: Ubuntu 22.04 x64
- **ROS 2 Version**: Humble
- **Package Version**: `sick_scan_xd` v3.6 (latest release)
- **Hardware**: SICK LMS4000 Lidar
- **Launch Command**:
  ```sh
  ros2 run sick_scan_xd sick_generic_caller /root/ros2_ws/src/sick_scan_xd/launch/sick_lms_4xxx.launch
  ```

rostest, 2025-03-14:
Please use the following branch to avoid this problem:

https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx

asokolowski-it, 2025-03-18:
With this branch, I don’t have critical errors, but I still get some warnings (see below). Maybe this will help you during further development. Thank you for your assistance!

```
[INFO] [1742290184.410983225] [sick_scan_xd]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud_101"
[WARN] [1742293174.014686050] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp(u32Sec=1742288882, u32NanoSec=663175582, u64Tick=623): FirstTickScan=4294956261, Curtick-FirstTickScan=-4294955638, InterpolationSlope=9.99158e-07, (Curtick-FirstTickScan)*InterpolationSlope=1.84312e+13, relTimeStamp1=-4291.34, relTimeStamp2=-4291.34, invalid lidar ticks, extrapolated timestamp not ok.
[WARN] [1742293174.014720936] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp() failed, returning current system time 1742293174.014672756 [sec]
[WARN] [1742293174.014736831] [sick_scan_xd]: ## u64Curtick=623, FirstTickScan()=4294956261, Curtick-FirstTickScan=-4294955638, FirstTimeStamp()=1.74229e+09, InterpolationSlope()=9.99158e-07, corrTime=1.74229e+09
[WARN] [1742293174.016487005] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp(u32Sec=1742288882, u32NanoSec=664864063, u64Tick=2313): FirstTickScan=4294956261, Curtick-FirstTickScan=-4294953948, InterpolationSlope=9.99158e-07, (Curtick-FirstTickScan)*InterpolationSlope=1.84312e+13, relTimeStamp1=-4291.34, relTimeStamp2=-4291.34, invalid lidar ticks, extrapolated timestamp not ok.
[WARN] [1742293174.016517446] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp() failed, returning current system time 1742293174.016468763 [sec]
[WARN] [1742293174.016531221] [sick_scan_xd]: ## u64Curtick=2313, FirstTickScan()=4294956261, Curtick-FirstTickScan=-4294953948, FirstTimeStamp()=1.74229e+09, InterpolationSlope()=9.99158e-07, corrTime=1.74229e+09
[WARN] [1742293174.018053915] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp(u32Sec=1742288882, u32NanoSec=666536569, u64Tick=3987): FirstTickScan=4294956261, Curtick-FirstTickScan=-4294952274, InterpolationSlope=9.99158e-07, (Curtick-FirstTickScan)*InterpolationSlope=1.84312e+13, relTimeStamp1=-4291.34, relTimeStamp2=-4291.34, invalid lidar ticks, extrapolated timestamp not ok.
[WARN] [1742293174.018147670] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp() failed, returning current system time 1742293174.018043280 [sec]
[WARN] [1742293174.018155578] [sick_scan_xd]: ## u64Curtick=3987, FirstTickScan()=4294956261, Curtick-FirstTickScan=-4294952274, FirstTimeStamp()=1.74229e+09, InterpolationSlope()=9.99158e-07, corrTime=1.74229e+09
[WARN] [1742293174.019750251] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp(u32Sec=1742288882, u32NanoSec=668194293, u64Tick=5646): FirstTickScan=4294956261, Curtick-FirstTickScan=-4294950615, InterpolationSlope=9.99158e-07, (Curtick-FirstTickScan)*InterpolationSlope=1.84312e+13, relTimeStamp1=-4291.33, relTimeStamp2=-4291.33, invalid lidar ticks, extrapolated timestamp not ok.
[WARN] [1742293174.019759612] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp() failed, returning current system time 1742293174.019744873 [sec]
[WARN] [1742293174.019765437] [sick_scan_xd]: ## u64Curtick=5646, FirstTickScan()=4294956261, Curtick-FirstTickScan=-4294950615, FirstTimeStamp()=1.74229e+09, InterpolationSlope()=9.99158e-07, corrTime=1.74229e+09
[WARN] [1742293174.021373407] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp(u32Sec=1742288882, u32NanoSec=669866800, u64Tick=7320): FirstTickScan=4294956261, Curtick-FirstTickScan=-4294948941, InterpolationSlope=9.99158e-07, (Curtick-FirstTickScan)*InterpolationSlope=1.84312e+13, relTimeStamp1=-4291.33, relTimeStamp2=-4291.33, invalid lidar ticks, extrapolated timestamp not ok.
[WARN] [1742293174.021382282] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp() failed, returning current system time 1742293174.021367311 [sec]
[WARN] [1742293174.021388349] [sick_scan_xd]: ## u64Curtick=7320, FirstTickScan()=4294956261, Curtick-FirstTickScan=-4294948941, FirstTimeStamp()=1.74229e+09, InterpolationSlope()=9.99158e-07, corrTime=1.74229e+09
[WARN] [1742293174.023071844] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp(u32Sec=1742288882, u32NanoSec=671524524, u64Tick=8979): FirstTickScan=4294956261, Curtick-FirstTickScan=-4294947282, InterpolationSlope=9.99158e-07, (Curtick-FirstTickScan)*InterpolationSlope=1.84312e+13, relTimeStamp1=-4291.33, relTimeStamp2=-4291.33, invalid lidar ticks, extrapolated timestamp not ok.
[WARN] [1742293174.023083546] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp() failed, returning current system time 1742293174.023065090 [sec]
[WARN] [1742293174.023089128] [sick_scan_xd]: ## u64Curtick=8979, FirstTickScan()=4294956261, Curtick-FirstTickScan=-4294947282, FirstTimeStamp()=1.74229e+09, InterpolationSlope()=9.99158e-07, corrTime=1.74229e+09
[WARN] [1742293174.024691628] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp(u32Sec=1742288882, u32NanoSec=673166990, u64Tick=10623): FirstTickScan=4294956261, Curtick-FirstTickScan=-4294945638, InterpolationSlope=9.99158e-07, (Curtick-FirstTickScan)*InterpolationSlope=1.84312e+13, relTimeStamp1=-4291.33, relTimeStamp2=-4291.33, invalid lidar ticks, extrapolated timestamp not ok.
[WARN] [1742293174.024702724] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp() failed, returning current system time 1742293174.024683237 [sec]
[... cut ...] ionSlope=1.84312e+13, relTimeStamp1=-4291.31, relTimeStamp2=-4291.31, invalid lidar ticks, extrapolated timestamp not ok.
[WARN] [1742293174.044735351] [sick_scan_xd]: ## WARNING SoftwarePLL::getCorrectedTimeStamp() failed, returning current system time 1742293174.044670343 [sec]
[WARN] [1742293174.044750024] [sick_scan_xd]: ## u64Curtick=30621, FirstTickScan()=4294956261, Curtick-FirstTickScan=-4294925640, FirstTimeStamp()=1.74229e+09, InterpolationSlope()=9.99158e-07, corrTime=1.74229e+09

```

closed 2025-03-18 by asokolowski-it

csh614, 2025-07-30:
I have same error code.

terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot store a negative time point in rclcpp::Time

Operating System: Ubuntu 20.04 x64
ROS 2 Version: Humble
Hardware: SICK Tim 571 Lidar