SICKAG/sick_scan_xd #462: Cannot get PTP to work with issue_update_2412xx branch

Navsea, 2025-03-20:
Hi,

I saw that I needed to use this updated branch to make PTP work.
But even with that update, I cannot seem to get a sychronized timestamp in the point cloud header ros2 message.
I tested with another PTP enabled device (a camera) and that seems to work fine.

I'm using the multiscan 136 and an nvidia jetson xavier platform (integrated by syslogic).

I attached the lidar launch file, where you can see that the timestamp mode is set to 2
[image]

I have set the TSCRole setting to PTP: 
[image]

Checking the PTP protocol packets, I see that the lidar and master is communicating, but it seems that the lidar is reporting an empty originTimestamp:
[image]

The PTP master does have a receiveTimestamp filled in:
[image]

When using the PTP enabled camera, I do see that the camera has filled in the values for originTimestamp:
[image]

Any help with this would be much appreciated.
We're working on getting the sensor off the vehicle and on a testbench, which I will be able to access remotely.
We could even have a session together if desired.
Will try to get firmware version as soon the vehicle comes online again (it's with a customer)

Best regards,
Kenneth De Leener

Navsea, 2025-03-21:
Update: 
I just checked the firmware vesion

[image]

reinbja, 2025-03-24:
Hi Kenneth,
Please update the sensor to the latest firmware 2.3.0. You will find the firmware in the download register of the multiScan https://www.sick.com/ag/en/catalog/products/lidar-and-radar-sensors/lidar-sensors/multiscan100/muls1aa-112211-multiscan136/p/p680032?category=g569793&tab=downloads

To activate PTP go to register configuration -> basic-settings

[image]

Navsea, 2025-03-25:
Hi Reinbja,

Thanks for the quick repsonse.
I was using E2E and did not specify 1 or 2 step.
In the manual it says that therefore it should default to 2-step mode. (https://manpages.ubuntu.com/manpages/bionic/man8/ptp4l.8.html)
I do have a log of the ethernet traffic if you'd like.
I only saw delay_req messags coming from the lidar.

Once we have the sensor back and on our bench, I will update to the latest firmware.
Thanks for the help!

reinbja, 2025-03-26:
Hi @Navsea 
If there is a .pcap file of the traffic, please attach it to the ticket, and we will look at it.

boehrluSICKAG, 2025-03-27:
Hi @Navsea ,
thanks for adding the information. I have checked the .pcap file, it is exactly as you have described. Nothing more to add I would say.
Regarding your question of "originTimestamp" beeing 0: This is not necessarily an issue, the originTimestamp only is one part of the PTP delay request process. It only tells when the scanner actually sent the delay_req message. In the general PTP context this shouldnt be an issue, the synchronization should still work as expected.
As soon as you have the scanner on your workbench you can verify this by activating PTP on the webserver (as mentioned by @reinbja ) and connect the device to the respective PTP master. After a few seconds (and a reload of the webserver) you can see the updated time on the device interface below the PTP settings.

Now getting to your issue with the ROS timestamp, where did you change the tick mode? Just to make sure you are working in /<Your sick_scan_xd Workspace>/install/sick_scan_xd/share/sick_scan_xd/launch/sick_multiscan.launch
I could not reproduce your issue. I am using the branch containing the fix 2412xx. I have tested with the newest device firmware (2.3.0), but will do some further tests with 2.0.1.

Can you please come back to me whenever you had time to check the mentioned parts? Any hints that could help me reproduce this issue is highly appreciated.

Thanks & best regards,
Lukas

Navsea, 2025-04-02:
Hi all,

So the tick mode was in fact set correctly in the launch file located in the install folder.

We first updated the firmware to the latest version (v2.3.0), and it still didn't work.
We unmounted it and tried on a test bench, with a clean image.
There it did work.

I then copied over the complete driver folder onto the vehicle controller, and then it did work.
So it seems that i might have done something wrong in downloading the driver, or configuration...
We did not have direct access to git on the vehicle, so i simply downloaded the feature branch as a zip file, and used that inside the sick_scan workspace.

On the bench i simply cloned the sick_scan_xd repository and checked out the feature branch. 
(without libsick_ldmrs which is mentioned here: https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx#build-on-linux-ros-2).

Anyhow, thank you for the assistance. 
If I find any conclusive reasons, i'll be sure to let you know.

Best regards,
Kenneth

boehrluSICKAG, 2025-04-02:
Hi Kenneth,

I can't really tell why it didn't work via zipped feature branch, it should also work.
But the main thing is that it worked out for you, great to hear.

If you have any follow-up (or new) questions feel free to reach out to us.

Best regards,
Lukas