SICKAG/sick_scan_xd #465: SICK_ldmrs no such file or directory

Nervxz, 2025-03-26:
Hi, 
i want to ask that when i run the `colcon build --cmake-args " -DROS_VERSION=2" --event-handlers console_direct+` in the link [https://github.com/SICKAG/sick_lidar_localization](url) for ROS 2 i face this issues:

`[ 76%] Built target sick_scan_xd_s__rosidl_typesupport_c`
`[Processing: sick_scan_xd]`                             
`gmake[1]: *** [CMakeFiles/Makefile2:630: CMakeFiles/sick_scan_xd_lib.dir/all] Error 2`
`gmake[1]: *** Waiting for unfinished jobs....`
`gmake[1]: *** [CMakeFiles/Makefile2:658: CMakeFiles/sick_scan_xd_shared_lib.dir/all] Error 2`
`gmake: *** [Makefile:146: all] Error 2`
`--- stderr: sick_scan_xd`
`In file included from /home/nervx/sick_scan_ws/src/sick_scan_xd/include/sick_scan/ldmrs/sick_ldmrs_node.h:12,`
 `                from /home/nervx/sick_scan_ws/src/sick_scan_xd/driver/src/sick_generic_laser.cpp:72:`
`/home/nervx/sick_scan_ws/src/sick_scan_xd/include/sick_scan/ldmrs/sick_ldmrs_driver.hpp:58:10: fatal error: sick_ldmrs/manager.hpp: No such file or directory`
 `  58 | #include <sick_ldmrs/manager.hpp>`
`compilation terminated.`
`In file included from /home/nervx/sick_scan_ws/src/sick_scan_xd/include/sick_scan/ldmrs/sick_ldmrs_node.h:12,`
 `                from /home/nervx/sick_scan_ws/src/sick_scan_xd/driver/src/sick_generic_laser.cpp:72:`
`/home/nervx/sick_scan_ws/src/sick_scan_xd/include/sick_scan/ldmrs/sick_ldmrs_driver.hpp:58:10: fatal error: sick_ldmrs/manager.hpp: No such file or directory`
`   58 | #include <sick_ldmrs/manager.hpp>`
`compilation terminated.`
`gmake[2]: *** [CMakeFiles/sick_scan_xd_lib.dir/build.make:216: CMakeFiles/sick_scan_xd_lib.dir/driver/src/sick_generic_laser.cpp.o] Error 1`
`gmake[2]: *** Waiting for unfinished jobs....`
`gmake[2]: *** [CMakeFiles/sick_scan_xd_shared_lib.dir/build.make:216: CMakeFiles/sick_scan_xd_shared_lib.dir/driver/src/`sick_generic_laser.cpp.o] Error 1`
`gmake[2]: *** Waiting for unfinished jobs....
`gmake[1]: *** [CMakeFiles/Makefile2:630: CMakeFiles/sick_scan_xd_lib.dir/all] Error 2`
`gmake[1]: *** Waiting for unfinished jobs....`
`gmake[1]: *** [CMakeFiles/Makefile2:658: CMakeFiles/sick_scan_xd_shared_lib.dir/all] Error 2`
`gmake: *** [Makefile:146: all] Error 2`
`Failed   <<< sick_scan_xd [50.1s, exited with code 2]`
`Summary: 2 packages finished [50.3s]`
 ` 1 package failed: sick_scan_xd`
 ` 1 package had stderr output: sick_scan_xd`

My LiDAR is Multiscan 136 1131164

weinmalSICKAG, 2025-03-26:
Hello,
thank you for your message.
From the [Readme](https://github.com/SICKAG/sick_scan_xd?tab=readme-ov-file#build-from-sources-1) you can see that if you don't need LD-MRS support, you can skip it. Use the following command to build without LD-MRS support.
```
colcon build --packages-select sick_scan_xd --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" --event-handlers console_direct+
```

Nervxz, 2025-03-26:
thank you for your response
by the way, i want to received LiDAR multiscan 136 data and make it into visualize data, how can i do it. i have run `ros2 launch sick_scan_xd sick_multiscan.launch.py hostname:=<ip-address> udp_receiver_ip:=<ip-address>` and it keep running, sorry it quite challenge for me because i have this LiDAR for sponsorship i have a lots questions with it

weinmalSICKAG, 2025-03-26:
No problem. You can use `rviz` to visualize the data: https://docs.ros.org/en/jazzy/Tutorials/Intermediate/RViz/RViz-User-Guide/RViz-User-Guide.html

Nervxz, 2025-03-26:
thank you very much
if i have any further question can i contact you by any chance in some ways ?

weinmalSICKAG, 2025-03-26:
If you have further questions regarding sick_scan_xd you are welcome to open another issue. For all other matters please contact SICK support at https://support.sick.com/.

Nervxz, 2025-03-26:
thank you very much, wish you have a nice day

closed 2025-03-26 by Nervxz