SICKAG/sick_scan_xd #467: Timing issues - multiScan 165

Kai-Glz, 2025-03-26:
Hi, 
I use the multiScan 165 with ROS2 Jazzy. 

It happens that the timestamp becomes negative when “tick_to_timestamp_mode” = 0. It is sporadic and I can't really reproduce it. 
This option seems to assume that the sick scanner and the system clock run at exactly the same speed. If the system clock jumps or runs a few microseconds faster or slower than the sick clockticks, it would deviate over time.

If “tick_to_timestamp” is configured to 1 or 2, the timestamp remains at 0. 

The same with the branch "Commit e213f18".

Another question: Is the sensor also able to support PTP, as described in ticket (https://github.com/SICKAG/sick_scan_xd/issues/462) for the other Multiscan?

Many thanks in advance!

Kind regards, Kai

reinbja, 2025-03-26:
Hi Kai,
Which firmware is running on the sensor, at the moment? The latest firmware version is V2.3.0. You can check the firmware via REST. REST command http://**_your IP address_**/api/FirmwareVersion

The latest firmware can be downloaded [here](https://www.sick.com/ag/en/catalog/products/lidar-and-radar-sensors/lidar-sensors/multiscan100/muls1aa-112211-multiscan136/p/p680032?category=g569793&tab=downloads)

Since you are using branch [feature/issue_update_2412xx](https://github.com/SICKAG/sick_scan_xd/compare/feature/issue_update_2412xx), you should set up your launch file parameters as followed

Setup in launchfile [sick_multiScan.launch](https://urldefense.com/v3/__https://github.com/SICKAG/sick_scan_xd/blob/develop/launch/sick_multiscan.launch__;!!BXYwEpu2!toaf_jJqc3dgghlzMRMzMvI79uHqX6_c01GCtD-3jfVvvifcOyPVKmM33SGoVaWZAC4brF-V1MAh_jk$)

set parameter    [tick_to_timestamp_mode" type="int" value="2"](https://urldefense.com/v3/__https://github.com/SICKAG/sick_scan_xd/blob/8e560a865ea3642bb0230183b71ac149219978dd/launch/sick_multiscan.launch*L284__;Iw!!BXYwEpu2!toaf_jJqc3dgghlzMRMzMvI79uHqX6_c01GCtD-3jfVvvifcOyPVKmM33SGoVaWZAC4brF-V4EJzeJs$) 
use field parameter ["t"](https://urldefense.com/v3/__https://github.com/SICKAG/sick_scan_xd/blob/8e560a865ea3642bb0230183b71ac149219978dd/launch/sick_multiscan.launch*L143__;Iw!!BXYwEpu2!toaf_jJqc3dgghlzMRMzMvI79uHqX6_c01GCtD-3jfVvvifcOyPVKmM33SGoVaWZAC4brF-VaD-Yrf4$) (time offset in nano seconds relative to the header timestamp in the point cloud (4 byte, uint32), used by rtabmap for deskewing )

In addition,
If you have a .pcap-file of your ETH-stream, this could also be helpful for better investigation.

Best Regards 
Jan

Kai-Glz, 2025-03-26:
Thanks for the quick response.
I've updated the sensor and tried to set the ptp option.

Am I right in assuming that I have to change the configuration and then save it at the top of the bar?
When saving, I get the response java.util.concurrent.ExecutionException:de.sick.soas.device.api.DAAccessException (/api/SetLastUser)

I logged in and didn't change the password. Userlevel: Service

If I set the sensor to PTP, does the parameter "tick_to_timestamp_mode" still have an influence?

After setting the LiDAR to PTP I will try the stated parameters.

Thanks!

reinbja, 2025-03-26:
Yes, you need to save the parameter after you have changed s.th. For this, press the **disc icon**. After that the icon shall change it`s color from yellow (parameter changed and not saved) to green (parameter saved) 

[image]

If the parameter tick_to_timestamp_mode = 2, the sensor specific time (PTP) will be taken for the header.

Kai-Glz, 2025-03-31:
Hi, it seems to work thank you.

An additional question: 
Currently, the lidar and the IMU transmit with the same frame ID. Can I change the IMU frame ID somehow? 
I know it doesn't necessarily fit into the ticket. I can open a new one if you want me to.

Thanks!

weinmalSICKAG, 2025-03-31:
Please open a separate issue.

closed 2025-03-31 by weinmalSICKAG