SICKAG/sick_scan_xd #469: how to send SOPAS command picoscan 150 NO ROS

sisusch, 2025-03-27:
Hello everyone,

I'm using the c++ library (not ros), the .launch configuration file is loaded without problems and also the callback with which I take the points works. Now what I would like to do, since the library being a monolith designed for ROS, does not allow me, is to modify a specific parameter of the sensor, not included in the configuration file. To do this, as I learned, I can use the sopas commands, in particular sEN LIDinputstate 1 (picture below)

[image]

Only to do that I have to wait for the sensor initialization process, through the SickScanApiInitByCli(apiHandle, 2, sick_scan_param) function to have opened the TCP port etc., otherwise I get errors. My question, is there a better method than the one described above? If not, how can I verify that the initialization process has concluded?

P.S. I have currently inserted a 30s delay, to give the initialization procedure time to complete, but it is not the solution I would like.

rostest, 2025-04-06:
At the end of a successful initialisation process, the measurement mode is activated and the lidar sends its sensor data. It is recommended that a "sensor ready flag" is set when the first point cloud message is received. Once this flag is set, the sensor has been successfully initialised and it is safe to query sensor states using SOPAS commands.

rostest, 2025-04-09:
Clarified due to proposed solution.

closed 2025-04-09 by rostest