SICKAG/sick_scan_xd #470: NO TRIGGER callback for SickScanLIDoutputstateMsg messages NO ROS

sisusch, 2025-03-27:
hI

I would like to use the callback provided by the wrapper.c present in the library. I am running a program in c++ (no ros) using the methods provided by sick_scan_xd_api_wrapper. (I imported it as is without modifying it).
Now this below is how I called the method to register the callback.

```
void custom_output(SickScanApiHandle apiHandle, const SickScanLIDoutputstateMsg* msg) {

	std::cerr << "Erro register call back  " << LidarProcIdStr << "!" << std::endl;
}
```

```
	// Register call back to read status of DO
	int reg_io_result = SickScanApiRegisterLIDoutputstateMsg(sharedData->Lidar[LidarProcId].apiHandle, &custom_output);
	if (reg_io_result != 0) {
		std::cerr << "Erro register call back  " << LidarProcIdStr << "!" << std::endl;
		return -1;
	}
```

By placing a breakpoint inside the callback it is never triggered, Does anyone of you know what the cause could be?

Below is the web interface where the port 2 out count is incremented, s

[image]

rostest, 2025-04-03:
Can you record via WireShark the output traffic during change of the output state and provide this data?
The "LIDoutputstate" command signals the change in this IP traffic.
See https://cdn.sick.com/media/docs/1/91/691/operating_instructions_picoscan150_2d_lidar_sensors_en_im0106691.pdf for further background information.

rostest, 2025-04-03:
We guess, that you are talking about picoScan150 (like in your former tickets).

sisusch, 2025-04-04:
Hi Rostest,

Yes, the lidar I'm working on is the Picoscan 150.
Attached are the files with the traffic acquisition, the configuration file and the CLI response of the program.

From what I see there is no traffic from the sensor on port 2111 when I change the state of the digital outputs.

The above traffic acquisition was done without the web interface open, to minimize network traffic.
The strange thing is, if I verify the change of state of the pin, with the web graphic interface, this intercepts it (see photo in the first post), but no traffic, except for the initial configuration, is transmitted from the sensor port 2111. Below are the filtered messages.

Thanks for the support, regards.

[image]

rostest, 2025-04-06:
Thank you for your feedback. After further investigation, we can confirm the following situation:

Field monitoring and LIDoutputstate messages are currently only supported for LMS1xx, LMS5xx, TiM7xx and TiM7xxS, but not for picoScan devices and multiScan device. See https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx#field-evaluation-information for details. We will clarify a possible future support regarding LIDoutputstate messages for picoScan devices with SICK.

sisusch, 2025-04-07:
Thanks Rostest for the answer, I thought that being a new product the page was not completely updated, in any case now I have confirmation of the impossibility of using these extra fields for the picoscan family.

kind regards

closed 2025-04-07 by sisusch