SICKAG/sick_scan_xd #471: Frame ID LiDAR <-> IMU

Kai-Glz, 2025-03-31:
Currently, the lidar and the IMU transmit with the same frame ID. Can I change the IMU frame ID somehow?
The distances between the two coordinate systems are as described in the documentation:

[image]

Thanks!

rostest, 2025-04-03:
Hi @Kai-Glz,

Yes, the IMU frame ID is set in [`sick_generic_imu.cpp`](https://github.com/SICKAG/sick_scan_xd/blob/develop/driver/src/sick_generic_imu.cpp) via:

```cpp
imuMsg_.header.frame_id = commonPtr->config_.imu_frame_id;
``` 
To ensure it’s set correctly, define the parameter in your launch file like this. Make sure you're using the correct launch file—one way to confirm is by giving it a very specific name.

<param name="imu_frame_id" value="imu_link"/>

You can verify the parameter at runtime with:
```
rosparam get /your_node_namespace/imu_frame_id
```

Kai-Glz, 2025-04-07:
Hi @rostest ,

I added the line in the sick_multiscan.launch.
 <param name="imu_frame_id" type="string" value="sick_imu"/>

But nothing happens.

rostest, 2025-04-08:
Thanks for following up! For multiScan devices, the imu frame id is configured by launchfile parameter "publish_frame_id". We will add the parameter "imu_frame_id" for multiScan and picoScan devices in a future release.
Currently you have the following alternatives to change the IMU frame id:
* Set parameter `publish_frame_id` to the imu frame id in the launchfile (e.g. `<arg name="publish_frame_id" default="sick_imu" />`) or overwrite it by commandline argument `publish_frame_id:="sick_imu"`. Optionally change the `frameid` parameters for the pointclouds.
* Set the imu frame id in the file driver/src/sick_scansegment_xd/ros_msgpack_publisher.cpp line 922: `imu_msg.header.frame_id = "sick_imu"; // m_frame_id;`

closed 2025-04-08 by Kai-Glz