SICKAG/sick_scan_xd #474: Driver says good bye and stop when testing with pico150 prime

Jonny0816, 2025-04-09:
Dear Sir 

Please see below sreenshoot that customer feedback the driver response good bye and stop when testing with pico150 prime without any other telegram comunication. Please help to analysis what is the issue and how to solve or avaid it, Thanks.

[image]

rostest, 2025-04-09:
Hi,

please read the information given in https://github.com/SICKAG/sick_scan_xd/issues/372 
and follow the instructions given there.

nancy-zhang-hub, 2025-04-11:
supplementary information for this issue:
there is an error record form sick driver in system log before the good bye

Apr 09 09:43:30 phoenix102 kernel: traps: sick_generic_ca[1831] trap divide error ip:55d369073a0d sp:7f67977fc920 error:0 in sick_generic_caller[55d368ee3000+1e5000]

nancy-zhang-hub, 2025-04-11:
and, the error happen after robot has been running for a few days(normal running)
then driver error and say good bye while robot is running!!!
This doesn't make sense.

Jonny0816, 2025-04-15:
@rostest Please help to give some comments about this, the driver had been running for long time before it reported error and say good bye, it is not same with #272 about firewall.
Thanks.

rostest, 2025-04-15:
Hi,

the information given here is quite useful:

```
supplementary information for this issue:
there is an error record form sick driver in system log before the good bye

Apr 09 09:43:30 phoenix102 kernel: traps: sick_generic_ca[1831] trap divide error ip:55d369073a0d sp:7f67977fc920 error:0 in sick_generic_caller[55d368ee3000+1e5000]
```

In the breakdown it gives use the following information:

- **Date and time**: `Apr 09 09:43:30`
- **Hostname**: `phoenix102`
- **Process**: `sick_generic_ca[1831]`  
  This is likely short for `sick_generic_caller`, a process with PID 1831.
- **trap divide error**:  
  Indicates a CPU exception due to division by zero or a similar issue.
- **ip: 55d369073a0d**:  
  Instruction pointer where the error occurred (i.e., the exact CPU instruction address).
- **sp: 7f67977fc920**:  
  Stack pointer at the time of the crash.
- **error: 0**:  
  Standard value in this kind of trap; not very informative here.
- **in sick_generic_caller[55d368ee3000+1e5000]**:  
  This shows that the crash occurred inside the binary `sick_generic_caller`, which was loaded at memory address `55d368ee3000` and had a size of `1e5000` bytes (about 1.8 MB).

rostest, 2025-04-15:
Next steps:

1. Calculating the offset:
    ```
   OFFSET = IP - BASE = 55d369073a0d - 55d368ee3000 = 0x1a0a0d
   ```
   So the crash happened at offset 0x1a0a0d in the sick_generic_caller binary.

2. Providing the binary and build environment information.
Can you upload the binary somewhere to make it available for further analysis (e.g. zip it and attach it to the ticket).
Can you provide further information about the build environment?

3. Can you run the following command?
    ```
   objdump -d sick_generic_caller >objdump_out.txt 
   ```
   and provide the file `objdump_out.txt`?

rostest, 2025-04-15:
Additional information:

In addition, the driver can be built with debug symbols or started directly with gdb. However, since the error only occurs rarely and after several days, I would strongly suggest that we start with the **existing binary first**. Therefore, please do not rebuild, but keep the binary as it is and share it here.

nancy-zhang-hub, 2025-04-15:
yes
ubuntu20.04, ROS1, sick dirver 3.6.0
1.show the sick_generic_caller and the objdump_out.txt  here

tar -xzvf sick_scan_xd.tar.gz
2.show sick driver file here, use the launch file: sick_picoscan_150.launch

same changes in driver files
1.Comment out some logs.

[image]
2.close the debug mode 

[image]

we encounter the goog bye issue when start using picoscan150 and driver 3.6.0, and i revised the launch file and seems solved it 
then after many days, the new good bye coming, robot it running while this issue happen

rostest, 2025-04-15:
Hi, 

thanks for providing the exe and the disassembly.
We recalculate the address offset. This gives us:

[image]

The disassembling shows:

[image]

rostest, 2025-04-15:
We also recommend that you use the branch

**https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx**

which solves problems with time stamping. This may also solve your problem.

To isolate the error further, you could also start the driver with gdb and at least compile and link it with debug symbols.

nancy-zhang-hub, 2025-04-16:
1.show the launch file below,sick_picoscan_150.launch
@rostest .2.so we didn't found out the root reason and need change to another branch for testing or the reason is the time stamping
3. if i use the new branch, can i use the previous setting, comment out log, close the build debug target, and the launch file below
@rostest  

sick_picoscan_150.launch
<?xml version="1.0"?>
<launch>
    
    <arg name="nodename1" default="sick_picoscan1"/>
    <arg name="hostname1" default="192.168.0.1"/>
    <arg name="udp_receiver_ip1" default=""/>

    
    <arg name="nodename2" default="sick_picoscan2"/>
    <arg name="hostname2" default="192.168.0.2"/>
    <arg name="udp_receiver_ip2" default=""/>

    
    <arg name="publish_frame_id" default="cloud" />
    <arg name="scandataformat" default="2"/>
    <arg name="performanceprofilenumber" default="2"/>
    <arg name="all_segments_min_deg" default="-90.0" />
    <arg name="all_segments_max_deg" default="+90.0" />
    <arg name="host_FREchoFilter" default="0" />
    <arg name="custom_pointclouds" default="cloud_all_fields_fullframe"/>
    <arg name="add_transform_xyz_rpy" default="0,0,0,0,0,0"/>

    
    <node name="$(arg nodename1)" pkg="sick_scan_xd" type="sick_generic_caller" respawn="false" output="screen" required="true">
        <param name="scanner_type" type="string" value="sick_picoscan"/>
        
        <param name="hostname" type="string" value="$(arg hostname1)" />
        <param name="port" type="int" value="2115" />
        <param name="udp_receiver_ip" type="string" value="$(arg udp_receiver_ip1)" />
        
        <param name="udp_sender" type="string" value="" />
        <param name="udp_port" type="int" value="2115" />
        <param name="all_segments_min_deg" type="double" value="$(arg all_segments_min_deg)" />
        <param name="all_segments_max_deg" type="double" value="$(arg all_segments_max_deg)" />
        <param name="publish_frame_id" type="string" value="$(arg publish_frame_id)" />
        <param name="udp_input_fifolength" type="int" value="20" />
        <param name="msgpack_output_fifolength" type="int" value="20" />
        <param name="verbose_level" type="int" value="1" />
        <param name="measure_timing" type="bool" value="True" />
        <param name="export_csv" type="bool" value="False" />
        <param name="export_udp_msg" type="bool" value="False" />
        <param name="logfolder" type="string" value="" />
        <param name="send_udp_start" type="bool" value="False" />
        <param name="send_udp_start_string" type="string" value="magicalActivate" />
        <param name="udp_timeout_ms" type="int" value="60000" />
        <param name="scandataformat" type="int" value="$(arg scandataformat)" />
        <param name="performanceprofilenumber" type="int" value="$(arg performanceprofilenumber)"/>
        
        <param name="imu_enable" type="bool" value="False"/>
        
        <param name="add_transform_xyz_rpy" type="string" value="$(arg add_transform_xyz_rpy)" /> 
        
        <param name="start_sopas_service" type="bool" value="False" />
        
        <param name="host_read_filtersettings" type="bool" value="True" />
        <param name="host_FREchoFilter" type="int" value="$(arg host_FREchoFilter)" />
        <param name="host_set_FREchoFilter" type="bool" value="True" />
        
        <param name="msgpack_validator_enabled" type="bool" value="False" />
        
        <param name="laserscan_layer_filter" type="string" value="1" />
        
        <param name="custom_pointclouds" type="string" value="$(arg custom_pointclouds)"/>
        
[... 23 lines cut ...]
        <param name="send_udp_start" type="bool" value="False" />
        <param name="send_udp_start_string" type="string" value="magicalActivate" />
        <param name="udp_timeout_ms" type="int" value="60000" />
        <param name="scandataformat" type="int" value="$(arg scandataformat)" />
        <param name="performanceprofilenumber" type="int" value="$(arg performanceprofilenumber)"/>
        
        <param name="imu_enable" type="bool" value="False"/>
        
        <param name="add_transform_xyz_rpy" type="string" value="$(arg add_transform_xyz_rpy)" /> 
        
        <param name="start_sopas_service" type="bool" value="False" />
        
        <param name="host_read_filtersettings" type="bool" value="True" />
        <param name="host_FREchoFilter" type="int" value="$(arg host_FREchoFilter)" />
        <param name="host_set_FREchoFilter" type="bool" value="True" />
        
        <param name="msgpack_validator_enabled" type="bool" value="False" />
        
        <param name="laserscan_layer_filter" type="string" value="1" />
        
        <param name="custom_pointclouds" type="string" value="$(arg custom_pointclouds)"/>
        
        <param name="cloud_all_fields_fullframe" type="string" value="topic=/$(arg nodename2) frameid=$(arg publish_frame_id) publish=1"/>
    </node>
</launch>

rostest, 2025-04-16:
Hi,

Thank you for your questions and comments. Please find our responses below:

 * Launch file syntax: You can use the existing launch files with the feature branch [feature/issue_update_2412xx](https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx) without modifications.

 * Previous settings: Your current configuration should remain compatible with this branch.

 * Driver vs. picoScan150 issue: As the reported error cannot be reproduced on our side, we are unable to draw a definitive conclusion. We recommend building the driver with debug symbols and, ideally, running it within gdb to gain more insight.

 * Timestamping: There were known issues with timestamping using the software PLL in version 3.6.0. We recommend switching to the above-mentioned feature branch to address this.

nancy-zhang-hub, 2025-04-17:
[image]
how did you get this? form this file?

[image]
but why i get the this， different with you

[image]

nancy-zhang-hub, 2025-04-17:
[image]

rostest, 2025-04-17:
Good point. 

In the uploaded objdump_out.txt I found the location you mentioned. Since this position did not make sense, I repeated the process with the attached exe. This produced the results I submitted. Both outputs (the one you sent and the one I generated) unfortunately give no indication of a ‘division by zero’ problem.

rostest, 2025-04-19:
Thanks for your update — here are some comments from our side:

- The kernel log message containing `"trap divide error"` clearly points to a division-related exception, likely an integer divide-by-zero.
- The timestamp in the kernel log matches the one in your provided screenshot (see below).
- However, the screenshot itself does **not** show a crash. Instead, it shows that **signal 2 (SIGINT)** was caught:
  - This indicates a clean interruption (e.g. Ctrl+C or `kill -2`)
  - Handled by the existing `rosSignalHandler`
  - Resulting in a graceful shutdown of `sick_generic_caller`
- This seems to **contradict** the kernel message — is it possible that someone manually terminated the process around the same time?
- Also, the log right before the “Good bye” message suggests that the measurement process had just been started or restarted.
- Do you have any insights into that? Could there have been two different events occurring close together?

Finally, regarding your remark:

> "the existing launch files with the feature branch [feature/issue_update_2412xx] or 3.6.0 without modifications will cause more 'good bye'"

Could you clarify why you expect this behavior?  
Have you observed unexpected shutdowns directly with these launch files, or is this a suspicion based on past experience?

[image]

nancy-zhang-hub, 2025-04-22:
well, seems can;t access to github, then turn to the email
is it possible that someone manually terminated the process around the same time?__no possible, robot is moving

and the googbye in sick_generic_caller(3.4.0 and 3.6.0), we experienced were:
for picoscan_150:
i used your launch file picoscan for picoscan_150, with this line, will goodbye every time after reboot (usually power down for a long time)
<param name="cloud_all_fields_fullframe" type="string" value="coordinateNotation=3 updateMethod=0 fields=x,y,z,i,range,azimuth,elevation,t,ts,lidar_sec,lidar_nsec,ring,layer,echo,reflector echos=0,1,2 layers=1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16 reflectors=0,1 infringed=0,1 rangeFilter=0,999,0 topic=/$(arg nodename1) frameid=world publish=1"/>
if i replace this line with, then the problem disappered, just omit some parameters.
<param name="cloud_all_fields_fullframe" type="string" value="topic=/$(arg nodename1) frameid=world publish=1"/>
for tim240
your factory default setting in tim240 is not binary, and so as your launch file in driver 2.0
but in dirver 3.4 and 3.6, the default setting in launch file (tim240) is binary, different with the setting in sensors, then, also will have good bye issue after every reboot,
the parameter in luanch file (3.4 and 3.6)<param name="use_binary_protocol" type="bool" value="true"/>
will, just set the binary false, same to the setting in sensors, goodbye will disappered

and i think, every question will trigger the goodbye signal, or your driver will turn to goodbye bye

image.png (view on web)

image.png (view on web)

nancy-zhang-hub, 2025-04-22:
and i think the goodbye is the result, not the reason

rostest, 2025-04-22:
Hi,

 * Can you provide your patched source code and the startup parameter for sick_scan_xd?
   We would like to compile this code on Ubuntu 20.04 and try to run the startup you provide.
 * We assume that you are running two lidars and that the second lidar is causing the problems. Correct?
 * Are you using the launch file provided in the chat? If you have no problem providing the launch file, that would be great.

rostest, 2025-05-13:
For further investigations ("Driver says Good Bye") we recommend the following steps:

*   **Compile and link with full debug information** by using the following command line option:
    ```bash
    catkin_make_isolated --install --cmake-args -DROS_VERSION=1 -DLDMRS=0 -Wno-dev -DO=0
    ```
    The generated `sick_generic_caller` can be used by GDB. The option `-DO=0` disables optimization and integrates the debug symbols into the binary.

    **Remark:**
    The line
    ```xml
    <node name="$(arg nodename)" pkg="sick_scan_xd" type="sick_generic_caller" respawn="false" output="screen" required="true" launch-prefix="gdb -ex run -ex bt --args" >
    ```
    is a replacement for the line
    ```xml
    <node name="$(arg nodename)" pkg="sick_scan_xd" type="sick_generic_caller" respawn="false" output="screen" required="true">
    ```
    You can use this `gdb` call also for other lidars like the `TiM240`.

*   **Launch the command** (of course change the IP addresses to your setup):
    ```bash
    roslaunch sick_scan_xd sick_picoscan_gdb.launch hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.100
    ```
    `sick_picoscan_gdb.launch` can be found in the attached zip file.

*   If the driver crashes and you kill it manually, you will see the backtrace.

*   Post this backtrace (or a screenshot) in case of a crash.

**Remark (Compilation Speed/Performance):**
If the debug version of the driver is too slow, replace the Option `-DO=0` with `-DO=2` in the `catkin_make_isolated` command. In this case, the driver is a release version with debug symbols integrated. The output from GDB might be harder to interpret compared to the full debug version (built with `-DO=0`) but can still be very useful.

[sick_picoscan_gdb.zip](https://github.com/user-attachments/files/20192126/sick_picoscan_gdb.zip)

rostest, 2025-05-13:
*   **Logging Verbosity:**
    If the setting `<param name='rosconsole_logger_level' value='info'/>` might produce too much information for you, consider adjusting this to `warn` or `error` if the `info` level output is too much.

rostest, 2025-06-23:
As the source of the error is not clear and the error cannot be reproduced and it is unclear whether it may be due to the middleware, we will close the ticket. If you have more recent findings, we will open it again.